motioneye-project/motioneye #2915: aarch64 docker?!?

mallory98e, 2024-01-17:
I'm still using the armhf docker images on many of my raspberry PIs.

https://hub.docker.com/r/ccrisan/motioneye/tags

With the PI5 now we are going on 64 bit (aarch64) for good.

With the latest motioneye in it, the current one which is 4 years old using:

motionEye Version	0.42.1
Motion Version	4.2.2
OS Version	Debian 9.12

```
Unable to find image 'ccrisan/motioneye:master-armhf' locally
master-armhf: Pulling from ccrisan/motioneye
eaaa43e22ddc: Pull complete 
38b5f9df17bb: Pull complete 
5a36f2cd0f41: Pull complete 
957aa23b13ac: Extracting [========================================>          ]  84.67MB/104.6MB
957aa23b13ac: Pull complete 
1b23938bf6aa: Pull complete 
Digest: sha256:1d7abea3e92e8cc3de1072a0c102a9d91ef2aea2a285d7863b1a5dadd61a81e1
Status: Downloaded newer image for ccrisan/motioneye:master-armhf
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
```

Can someone create a new docker repo for:

**master-aarch64**

If not can you provide instructions how to create the docker image for 64 bit?!

Thank you.

zagrim, 2024-01-19:
There's image for arm64 here: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye
And that's anyway the hugely more recent codebase, currently in pre-release stage. What I don't really know is how much those images have been tested, and if there might be some issues with them, but please try it out.

mallory98e, 2024-02-03:
Yeah your right it's a lot newer:

motionEye Version 	0.43.1b1
Motion Version 	4.6.0
OS Version 	Linux 6.1.69_1

Vs the old one:

motionEye Version 	0.42.1
Motion Version 	4.2.2
OS Version 	Debian 9.12

Well one thing with this new version I can notice that there are way more video saving formats however I having a strange issue with it, the image from the camera is kind of squeezed on a weird way like you would step on the image from the right. 

The direct feed from another motioneye looks good. There isn't many option for this :/

About stability, it's running since couple of days for me no other issues asides from this.

zagrim, 2024-02-04:
What sort of camera? Which video format? Have you tried other formats? Can you share an example screenshot of the distortion (even if party redacted)?

mallory98e, 2024-02-09:
Nothing to do with the movies or the recording I think it's a layout issue, it displays the image weirdly even in small square or in fullscreen it's obvious side by side next to the camera stream in the browser.

It is another Raspberry PI with the 8mp camera running motion, nothing extra and no I don't use any video codecs just MOVIE PASSTHROUGH this case....

```
daemon on
setup_mode off
log_level 5
log_type all
v4l2_palette 17
input -1
norm 0
frequency 0
rotate 0
width 1280
height 720
framerate 30
minimum_frame_time 0
netcam_keepalive off
netcam_tolerant_check off
auto_brightness off
brightness 0
contrast 0
saturation 0
hue 0
roundrobin_frames 1
roundrobin_skip 1
switchfilter off
threshold 1500
threshold_tune off
noise_level 32
noise_tune on
despeckle_filter EedDl
smart_mask_speed 0
lightswitch 0
minimum_motion_frames 1
pre_capture 0
post_capture 0
event_gap 60
max_movie_time 0
emulate_motion off
output_pictures off
output_debug_pictures off
quality 75
picture_type jpeg
ffmpeg_output_movies off
ffmpeg_output_debug_movies off
ffmpeg_timelapse 0
ffmpeg_timelapse_mode daily
ffmpeg_bps 500000
ffmpeg_variable_bitrate 0
ffmpeg_video_codec mpeg4
use_extpipe off
snapshot_interval 0
locate_motion_mode off
locate_motion_style box
text_right %Y-%m-%d\n%T-%q
text_changes off
text_event %Y%m%d%H%M%S
text_double on
snapshot_filename %v-%Y%m%d%H%M%S-snapshot
picture_filename %v-%Y%m%d%H%M%S-%q
movie_filename %v-%Y%m%d%H%M%S
timelapse_filename %Y%m%d-timelapse
ipv6_enabled off
stream_port 8081
stream_quality 50
stream_motion off
stream_maxrate 100
stream_localhost off
stream_limit 0
stream_auth_method 1
webcontrol_port 8082
webcontrol_localhost on
webcontrol_html_output on
track_type 0
track_auto off
track_iomojo_id 0
track_step_angle_x 10
track_step_angle_y 10
track_move_wait 10
track_speed 255
track_stepsize 40
quiet on
```

zagrim, 2024-02-10:
Oh, so you mean that the frame size is kind of compressed horizontally (or vertically)? I've had that same issue in my own setup for years, but I've never bothered to figure out why, since the recorded videos (using passthrough, too) are just fine. And I'm not running docker, so it's at least got nothing to do with the docker image.

There's this issue that might be related (depending on your stream type): #2693
Otherwise please open a new issue for the aspect ratio issue to not mix up several topics in this one.

MichaIng, 2025-04-28:
Sorry it seems I broke it by adding RISC-V images. But pushing with two workflow steps to the same tag does not seem to work. Merging them now: #3133

MichaIng, 2025-04-28:
That worked: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye
So latest `edge` tag now has images for all architectures (+RISC-V) now.

closed 2025-04-28 by MichaIng