motioneye-project/motioneye #2917: Camera not working

Gurpreet83, 2024-01-21:
Hi all,

how do I fix this please?

```
[0:17:35.754514819] [17203]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:17:35.808990353] [17206]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:17:35.811371913] [17206]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media0
[0:17:35.811468191] [17206]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Made X/EGL preview window
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[0:17:36.304410288] [17203]  INFO Camera camera.cpp:1183 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-SBGGR10_CSI2P
[0:17:36.304704993] [17206] ERROR V4L2 v4l2_videodevice.cpp:1047 /dev/video0[14:cap]: Unable to set format: Device or resource busy

ERROR: *** failed to configure streams ***
```

MichaIng, 2024-01-23:
If this is on a Raspberry Pi, and KMS is enabled (`dtoverlay=vc4-kms-v3d` in `/boot/config.txt`), then it fails expectedly. If it is a camera module 3 or newer, then it cannot work with the legacy camera interface either. However, here is a workaround: https://github.com/motioneye-project/motioneye/issues/2900#issuecomment-1902774793

adminy, 2024-03-11:
> Unable to set format: Device or resource busy

That means your camera is used by some other process.

JamBalaya56562, 2026-11-13:
motionEye 0.45.0 can use the camera module itself (#3546): with `libcamera_v4l2` set to the path of `v4l2-compat.so`, it starts Motion through libcamera's V4L2 layer, and `sudo /opt/motioneye/bin/motioneye_init --libcamera` sets this up. Until that release, the `libcamerify` steps in the wiki's [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack) page do the same. "Device or resource busy" means another program, e.g. `rpicam-hello`, still has the camera open. Closing.

closed 2026-11-13 by JamBalaya56562