motioneye-project/motioneye #2924: What is the plan to update Docker images?

TheBlueSky, 2024-02-04:
[Docker images](https://hub.docker.com/r/ccrisan/motioneye/tags) in Docker Hub is 3 years and 4 years old for dev and master respectively. What is the plan to update them? Are they bound to the [releases](https://github.com/motioneye-project/motioneye/releases)? If so, what is the plan for the releases? There have been plenty of changes since June 2020, but not a single release.

MichaIng, 2024-02-04:
Use these for now: https://github.com/motioneye-project/motioneye/pkgs/container/motioneye

When someone can help me to in case adjust our Docker workflow to assure images are properly marked as pre-releases/beta on Docker Hub, and how to assign a real version string, when we swap this, then we can do that right away for the `beta` branch: https://github.com/motioneye-project/motioneye/blob/beta/.github/workflows/docker.yml

My idea would be to keep `dev`/`edge` images here on the GitHub Docker container registry, and start uploading to Docker Hub when pushing to `beta` and later `main` branch, but with a proper version string and beta mark/suffix, instead of having only `edge` or similar as version string. Probably @kleini or @MarcelCoding can help with that?

TheBlueSky, 2024-02-04:
I can try doing this. Just let me know what the requirements are.

What I understood from your message is that you want something like:

| Branch | Publish to | Tags |
|---|---|---|
| dev | GitHub Container Registry | `dev`, `1.0-dev`, `1.0.13-dev` |
| beta | Docker Hub | `beta`, `1.0-beta`, `1.0.13-beta` |
| main | Docker Hub | `latest`, `1.0`, `1.0.13` | 

Is that correct?

Also, what other requirements there are?

MichaIng, 2024-02-04:
Jep, basically that. If possible, `0.43.1b1` would be great as beta version string, since this is the one we use/the schema we need to use for the Python package on PyPI. But if Docker Hub does not support suffixes like this, `-beta` or `-beta1` would be fine. Ideally of course taken from here: https://github.com/motioneye-project/motioneye/blob/beta/motioneye/__init__.py

And, and of course best would be to have the same workflow on all branches, hence doing the change on `dev` at first, and then assigning the tags based on the branch the push was done to. But it could be tricky, so a diversion is fine for now. Since we need to merge into one direction only, and this affects only the tag lines, it is not hard to maintain the difference.

TheBlueSky, 2024-02-04:
> If possible, 0.43.1b1 would be great as beta version string

If you are using tags for releases, I can use those tags as Docker image tags. So, the updated table

| Tag | Branch | Publish to | Tags |
|---|---|---|---|
| Any tag | dev | GitHub Container Registry | `edge` |
| 0.43.1b1 | beta | Docker Hub | `beta`, `0.43.1b1` |
| 0.43.1 | main | Docker Hub | `latest`, `0.43`, `0.43.1` | 
| 1.0.13 | main | Docker Hub | `latest`, `1`, `1.0`, `1.0.13` | 

[This](https://github.com/docker/metadata-action) GitHub action has the potential to do that. But I have not tried it yet.

It is worth noting that, especially for the `beta` and `main` branches, depends on how discipline we are to push the right tag to the right branch.

MichaIng, 2024-02-04:
Ah yes that looks great.

We use this `docker/metadata-action` in our workflow already. But I did not have a closer look into it, about how to pass the version tags to it, in an automated way.

And as a side note: Since I aim to create GitHub tags and releases for every update/push on `beta` and `main` banches. The upload could be triggered on (release) tags as well, instead of push to branches, and then the version string taken from the tag name, instead of from the `_init_.py`: https://github.com/motioneye-project/motioneye/releases
Just in case it is easier or even the way it is supposed to work.

> It is worth noting that, especially for the `beta` and `main` branches, depends on how discipline we are to push the right tag to the right branch.

Do you mean GitHub (release) tags? I intend to assign them strictly with `nN` suffix for `beta` branch and without for `main` branch.

Okay, now I see that it is indeed how it is working, mapping the GitHub tags to Docker Hub tags: https://github.com/docker/metadata-action?tab=readme-ov-file#basic

And now I see that in theory, our workflow should support this already, but indeed it has issues with the `b1` suffix: https://github.com/motioneye-project/motioneye/actions/runs/7315066439
> 0.43.1b1 is not a valid semver. More info: https://semver.org/

But we do not need the multiple major/minor/patch version tags, hence semver identification, but just using the tag as is should be fine. As said, it would be important that this is not seen as a stable release at Docker Hub then, but however marked as pre-release and not pulled automatically by Docker, unless explicitly defined. I am not sure whether/how the concept of beta/pre-releases works in Docker registries.

TheBlueSky, 2024-02-04:
> We use this docker/metadata-action in our workflow already.

Oh, yeah. I haven't gone through the entire file 😅

> The upload could be triggered on (release) tags as well, instead of push to branches, and then the version string taken from the tag name

Yes, I intend to remove the push to branches and use the tags instead.

> Do you mean GitHub (release) tags?

Yes. This needs to be controlled manually. Otherwise, we need to parse the version number and make sure it does not contain any suffix.

> And now I see that in theory, our workflow should support this already, but indeed it has issues with the b1 suffix:

We can fix this with a some condition that says "if it is `beta` branch, then use the `raw` type instead of `semver`".

MichaIng, 2024-02-05:
Currently if fails loudly, but effectively those failing tags are just skipped. So it would work to just add a `type=ref,event=tag` (if I see it correctly), to have a Docker tag generated from the GitHub tag: https://github.com/docker/metadata-action?tab=readme-ov-file#typeref
That would also replace the long semver type, and only the `0.43` and in some future a major version tag would need to be added via semver type.

TheBlueSky, 2024-02-05:
My idea was to remove the pull request and push from [here](https://github.com/motioneye-project/motioneye/blob/main/.github/workflows/docker.yml#L3) and replace them with tags. This will work if we are going to use tags for releases only.

Then when configuring metadata action, do something like `type=raw,value=edge,enable=${{ github.ref == format('refs/heads/{0}', 'dev') }}` and similar thing `beta` and `main` branches, taking into consideration their specific tags; for example `0.43.1b1` for `beta`, and semver and `latest` for main `main`.

MichaIng, 2024-02-05:
> My idea was to remove the pull request and push from [here](https://github.com/motioneye-project/motioneye/blob/main/.github/workflows/docker.yml#L3) and replace them with tags. This will work if we are going to use tags for releases only.

We need to keep PR and push to branch triggers for testing the Docker builds. But if I am not mistaken, the metadata action should be skipped if we are not going to upload anyway, just like the container registry login and upload. Though we might need to make the labels and tags in the `build-push-action` action conditional, otherwise the `${{ steps.meta.outputs.* }}` calls would fail.

Also it is good to have an `edge` image automatically created here on GitHub container registry on every push to `dev`, instead of needing to create/update a tag every time.

But for uploads to Docker Hub, it makes sense if they are only done on push to tag events.

> Then when configuring metadata action, do something like `type=raw,value=edge,enable=${{ github.ref == format('refs/heads/{0}', 'dev') }}` and similar thing `beta` and `main` branches, taking into [...]

As said, `type=ref,event=tag` seems to cover at least all long version string tags, like `0.43.1b1`, creating a Docker tag based on the GitHub tag, on tag events. But for the meta tags `edge`, `beta`, `latest` and `major.minor`+`major`, additional rules are required, of course. We could of course also create all those tags here on GitHub (all but `edge` at least), but this is an additional effort I am currently not keen to do. Should be possible to automate, but still needs to be done by someone.

dojomotion, 2024-04-04:
New to virtual environment.  Wondering if a Docker install is similar than venv install ? 
When i try to use the sequence proposed at  https://www.piwheels.org/project/motioneye/ , i get this error :
(venv_cam) pi@rpi115:~/prgs/venv_cam/bin $ meyectl startserver
Traceback (most recent call last):
  File "/home/pi/prgs/venv_cam/bin/meyectl", line 5, in <module>
    from motioneye.meyectl import main  File "/home/pi/prgs/venv_cam/lib/python3.9/site-packages/motioneye/meyectl.py", line 28, in <module>   import settings
ModuleNotFoundError: No module named 'settings'

(python 3.9 in my venv)

eyk107, 2025-02-21:
Any updates on this topic?