motioneye-project/motioneye #2933: How can we stream from camera to Motion server itself?

ttodua, 2024-02-23:
How can we feed video from device (i.e. IP camera running on Android) to the server itself? what I have seen, motioneye itself connects to the IP/stream, but can't we feed stream in opposite direction (from device to motioneye)? if so, please give some instructions

MichaIng, 2024-02-23:
You can add network cameras in motionEye. Hence if your Android is able to generate an RTSP/netcam/MJPEG stream and is reachable by motionEye at some IP or hostname, you can do it this way.

ttodua, 2024-02-24:
No, maybe I didn't express question well. 
I want in opposite. the phone is not in same network, however, it has got internet connection, so instead of MotionEye trying to connect it, i want in opposite - Android phone sent video stream to MotionEye server (so, motionEye had somewhat like "open" port or whatever, where it can accept the video stream coming from remote sources).

adminy, 2024-03-11:
Its the same thing ttodua,

Motioneye receives, camera sends, camera is an input device. Motioneye is an output device. What you want is probably port forwarding, but if that's too complicated, look into ngrok. Open an ngrok tcp port for 8765 and use the ngrok link to connect from your internet connected android from anywhere. (ngrok running on the same device as motioneye)

adminy, 2024-03-12:
Nope, if it did then it would be 2 problems instead of one. This is software to interact with the camera, that is software to manipulate the network. Very different things ;)

JamBalaya56562, 2026-11-12:
Thanks @adminy for the answer. motionEye connects to the camera's stream; it does not take a stream pushed to it. With the phone on another network, motionEye needs a way to reach the phone's stream, e.g. port forwarding, a VPN or a tunnel as suggested above. Closing as answered.

closed 2026-11-12 by JamBalaya56562