motioneye-project/motioneye #2934: MotionEye with Libcamera on Raspberry Pi 4B and Debian 10/11

Schnitzer50, 2024-02-28:
My installed Motioneye version is 22.4.1. Is there already a newer version? If so, what is the easiest way to install it?

zagrim, 2024-03-03:
Well, libcamera support in not that easy thing since Motion does not support it directly, and your other issue where you posted this same question (#2812) is just about that (although the issues I linked in the [second comment](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-1689734346) there are actually much more informative). There are some attempted workarounds, some of which seem to work for some people but not for all.

BTW, there exists no MotionEye version 22.4.1, so it's hard to answer your question on any newer version :smile: The newest version as seen in [releases page](https://github.com/motioneye-project/motioneye/releases) is 0.43.1b1 and the installation and upgrade instructions are here: https://github.com/motioneye-project/motioneye/blob/0.43.1b1/README.md

Schnitzer50, 2024-03-03:
﻿Thank you very much for the information. Attached are some screenshots of the Motioneye.Eo version.✉️  vom  📱Am 03.03.2024 um 17:34 schrieb Esa Tikka ***@***.***>:Well, libcamera support in not that easy thing since Motion does not support it directly, and your other issue where you posted this same question (#2812) is just about that (although the issues I linked in the second comment there are actually much more informative). There are some attempted workarounds, some of which seem to work for some people but not for all.
BTW, there exists no MotionEye version 22.4.1, so it's hard to answer your question on any newer version 😄 The newest version as seen in releases page is 0.43.1b1 and the installation and upgrade instructions are

zagrim, 2024-03-04:
I'm sorry but the attachments never made it to Github. I don't know if it's possible to attach files via email, but it's easy enough to do in the web interface.

Schnitzer50, 2024-03-04:
Here are the relevant attachments:

[image]

[image]

[image]

zagrim, 2024-03-12:
Your MotionEye is @jmichault's fork of this project somewhere at the time when this project still didn't have multi-lingual support. This is the corresponding Github project: https://github.com/jmichault/motioneye.eo and as can seen from the releases page the latest release is from [Jan 9, 2022](https://github.com/jmichault/motioneye.eo/releases/tag/22.1.5) which means it does not have any of the fixes and improvements made in the last two years to ME.

I recommend you to back up your configs and start from scratch with "the original" ME using these instructions:

> The newest version as seen in [releases page](https://github.com/motioneye-project/motioneye/releases) is 0.43.1b1 and the installation and upgrade instructions are here: https://github.com/motioneye-project/motioneye/blob/0.43.1b1/README.md

The config files are likely under `/etc/motioneye/`, back up at least all `.conf` files. Likely ME can also read them without messing up anything, but please do also read the [release notes](https://github.com/motioneye-project/motioneye/releases/tag/0.43.1b1).

JamBalaya56562, 2026-10-29:
As @zagrim found, 22.4.1 is the `motioneye.eo` fork, which has not been updated since January 2022. This project's current release is 0.44.0, installed as described in the README: https://github.com/motioneye-project/motioneye#installation. Raspberry Pi camera modules work through libcamerify for now, see https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack, and native libcamera support is tracked in #2812. Closing, as the question has been answered.

closed 2026-10-29 by JamBalaya56562