motioneye-project/motioneye #2936: Problème enregisterment MotionEye

FrThib, 2024-03-01:
Bonjour,
J'utilise MotionEye  sur mon Raspberry Pi4 4gb permettant la détection et l'enregistrement de vidéo capturée avec une caméra Avidsen ( mon réseau est constitué du Raspberry sur box en Ethernet, caméra en wifi sur box ) . Globalement cela  fonctionne bien (lecture, détection et enregistrement). Je suis néanmoins confronté à un problème que je ne comprends pas, il me manque des images et la progression dans le temps n'est pas cohérente.
Par exemple sur un test simple ou je présente devant la caméra ma main qui va décompter avec les doigts le temps ( 5 secondes ) la détection se fait correctement. Cependant lorsque je visualise le film je vois que partiellement le film et juste un passage furtif de ma main  
[image]
[image]
[image]

Avez-vous une idée pour ce problème et avez-vous comment le régler ?
Merci de votre retour et bonne année 2024
François

FrThib, 2024-03-04:
Hi evrybody
Nobody seems to be confronted to the similar problem I have, but maybe is due to the french language included in my message above.
I therefore reiterate my request in English
I use MotionEye on my Raspberry Pi4 4gb allowing the detection and recording of video captured with an Avidsen camera (my network consists of the Raspberry on box over Ethernet, camera over wifi on box). Overall it works well (reading, detection and recording). However, I am faced with a problem that I do not understand, I am missing images and the progression over time is not consistent.
For example, on a simple test where I present my hand in front of the camera which will count down the time with my fingers (5 seconds), the detection is done correctly. However when I view the film saved I only see partially the film and just a furtive passage of my handmain
Do you have any idea for this problem and how to fix it?
thank you for your comeback
Francois

zagrim, 2024-03-05:
You seem to have noise level ("Niveau de bruit") set quite high, and unless your camera operates in very low light, I suspect that the high noise level will cause some parts for video stream with lesser motion to be disregarded. So, I recommend lowering the noise level to 1% or lower and see if that helps.

I also recommend enabling "Show Frame Changes" ("Afficher les modifications d'image") so you can better see how the motion detection settings work. There will be a number on the top right corner showing the number of changed pixels in each frame. There's no need to raise noise level setting unless there are changed pixels even when there is no actual movement.

"Détection des changements de lumière" might also affect in a situation where most of the frame might change due to motion (I assume your hand might be close to the camera during the test). Having this set to 50% means that if more than 50% of frame pixels change, it might be interpreted as light conditions changing, and thus disregarded in motion detection. However, I'm not quite sure how that setting actually works in Motion so I might also be wrong here. It might however be helpful to either set that to 0% (disabling the feature) or closer to 100% to see if that affects your testing.

FrThib, 2024-03-05:
Many thanks zagrim for your help .
Indeed your settings impact my recording, however I have not yet managed to find the right setting to achieve the expected result. I will continue my investigations tomorrow. I also use “Show Frame Changes” as you asked, this allows me to better locate and understand the trigger zones.
thanks again
François

JamBalaya56562, 2026-11-12:
zagrim's advice is the way to tune this: a high noise level makes Motion ignore the frames with less change, so lower it and watch the changed pixels with "Show Frame Changes" while setting the threshold. As there was no follow-up since 2024, closing as answered; please open a new issue if frames are still missing with tuned settings.

closed 2026-11-12 by JamBalaya56562