motioneye-project/motioneye #2941: V4L2 device errors ambiguous and not showing

adminy, 2024-03-10:
I tried libcamera-vid, everything works, does not work in motioneye ui though ... same camera selected!
```js
Mar 10 17:31:27 DietPi meyectl[951]:     INFO: server started
Mar 10 17:31:34 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:31:38 DietPi meyectl[951]:     INFO: adding new v4l2 camera with id 1...
Mar 10 17:31:39 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:31:39 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:31:39 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:31:39 DietPi motion[981]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Mar 10 17:31:39 DietPi motion[981]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Mar 10 17:31:39 DietPi motion[981]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Mar 10 17:31:39 DietPi motion[981]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.6.0 Started
Mar 10 17:31:39 DietPi motion[981]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar 10 17:31:39 DietPi motion[981]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar 10 17:31:39 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:31:39 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:46:07 DietPi systemd[1]: Starting systemd-tmpfiles-clean.service - Cleanup of Temporary Directories...
Mar 10 17:46:07 DietPi systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
Mar 10 17:46:07 DietPi systemd[1]: Finished systemd-tmpfiles-clean.service - Cleanup of Temporary Directories.
Mar 10 17:46:07 DietPi systemd[1]: run-credentials-systemd\x2dtmpfiles\x2dclean.service.mount: Deactivated successfully.
Mar 10 17:49:56 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:50:01 DietPi meyectl[951]:     INFO: adding new v4l2 camera with id 2...
Mar 10 17:50:01 DietPi meyectl[951]:    ERROR: failed to list controls of device "/dev/media4"
Mar 10 17:50:03 DietPi motion[1088]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
Mar 10 17:50:03 DietPi motion[1088]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Mar 10 17:50:03 DietPi motion[1088]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
Mar 10 17:50:03 DietPi motion[1088]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Mar 10 17:50:03 DietPi motion[1088]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.6.0 Started
Mar 10 17:50:03 DietPi motion[1088]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Mar 10 17:50:03 DietPi motion[1088]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Mar 10 17:50:03 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:50:03 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:50:03 DietPi motion[1088]: [2:ml2:Camera2] [ERR] [VID] v4l2_device_capability: Not a V4L2 device?
Mar 10 17:50:03 DietPi motion[1088]: [2:ml2:Camera2] [ERR] [VID] vid_start: V4L2 device failed to open
Mar 10 17:50:03 DietPi motion[1088]: [2:ml2:Camera2] [WRN] [ALL] motion_init: Could not fetch initial image from camera
Mar 10 17:50:03 DietPi motion[1088]: [2:ml2:Camera2] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Mar 10 17:50:05 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:50:05 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:50:05 DietPi kernel: usb 1-1.2: Failed to query (GET_DEF) UVC control 2 on unit 1: -32 (exp. 1).
Mar 10 17:50:10 DietPi motion[1088]: [2:ml2:Camera2] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Mar 10 17:50:10 DietPi motion[1088]: [2:ml2:Camera2] [ERR] [VID] v4l2_device_capability: Not a V4L2 device?
Mar 10 17:50:10 DietPi motion[1088]: [2:ml2:Camera2] [ERR] [VID] vid_start: V4L2 device failed to open
Mar 10 17:50:20 DietPi motion[1088]: [2:ml2:Camera2] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Mar 10 17:50:20 DietPi motion[1088]: [2:ml2:Camera2] [ERR] [VID] v4l2_device_capability: Not a V4L2 device?
Mar 10 17:50:20 DietPi motion[1088]: [2:ml2:Camera2] [ERR] [VID] vid_start: V4L2 device failed to open
Mar 10 17:50:30 DietPi motion[1088]: [2:ml2:Camera2] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Mar 10 17:50:30 DietPi motion[1088]: [2:ml2:Camera2] [ERR] [VID] v4l2_device_capability: Not a V4L2 device?
Mar 10 17:50:30 DietPi motion[1088]: [2:ml2:Camera2] [ERR] [VID] vid_start: V4L2 device failed to open
```

zagrim, 2024-03-22:
Is this Pi camera 3? That requires libcamera support which is still somewhat open question for MotionEye. Motion requires using libcamerify in order to deal with those cameras, but integrating that to ME is still in progress. There have been some proposed workarounds, but some people have had issues with any of them. However, if you are feeling adventurous, see #2425 and the linked issues.

adminy, 2024-03-24:
I just got a defective camera it seems.

closed 2024-03-24 by adminy