motioneye-project/motioneye #2942: Not seeing Motion version in Settings

dog2bert, 2024-03-14:
I am not seeing the motion version.
Could this be why no camera's I add work?
I have installed motion while following this guide:
https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye

[image]

zagrim, 2024-03-17:
Please try installing using these instructions: https://github.com/motioneye-project/motioneye
The new pre-release of 0.43 should at least not require Python 2 and will support newer versions of Motion in case that might be the issue here.

dog2bert, 2024-03-18:
Is there a way to go back to a fresh install. Following your instructions it worked, but had many cameras where only one worked. I edited /etc/motion/motion.conf to remove cameras 2-4 and now the webinterface is broken. I also removed the other camera2-dist.conf files

zagrim, 2024-03-19:
I'm not sure how it should be best done, but if you delete (or move into a backup location) `/etc/motioneye/motion.conf` and any `/etc/motioneye/camera*.conf` files (but **do not** delete/move `/etc/motioneye/motioneye.conf` !), and then restart ME using `sudo systemctl restart motioneye`, you should be starting from a clean empty setup.

dog2bert, 2024-03-19:
Deleting and restarting didn't work, so I just reinstalled the OS, now it works

zagrim, 2024-03-20:
I'm sorry, but I have no experience on the quirks of USB cameras :disappointed: 

But if you attach logs showing MotionEye starting up and your camera config (redact any authentication related settings etc), maybe someone can help.
When you say "it shows 3", what is exactly shown in ME?

dog2bert, 2024-03-20:
Only one camera has output
[image]

dog2bert, 2024-03-20:
Where are the logs stored?

zagrim, 2024-03-21:
Logs can be seen using `sudo journalctl -u motioneye`. And please attach also the camera config files like I asked.

dog2bert, 2024-03-21:
[camera-1.txt](https://github.com/motioneye-project/motioneye/files/14705887/camera-1.txt)
[camera-2.txt](https://github.com/motioneye-project/motioneye/files/14705888/camera-2.txt)
[camera-3.txt](https://github.com/motioneye-project/motioneye/files/14705889/camera-3.txt)
[log.txt](https://github.com/motioneye-project/motioneye/files/14705890/log.txt)
[motion.txt](https://github.com/motioneye-project/motioneye/files/14705891/motion.txt)
[motioneye.txt](https://github.com/motioneye-project/motioneye/files/14705893/motioneye.txt)

zagrim, 2024-03-22:
Thanks. It seems that the error shown in the logs ("v4l2_input_select: Unable to query input 0. VIDIOC_ENUMINPUT, if you use a WEBCAM change input value in conf by -1: Inappropriate ioctl for device") is not that rare, and suggests that either one is not using the correct device file (generated by Linux kernel/udev) or that Linux kernel (or v4l2 subsystem) is having trouble setting up things for the camera.

From this project I found this one, which is somewhat depressing: https://github.com/motioneye-project/motioneye/issues/1934
If indeed there is an issue with Linux/V4L2 in general when having multiple USB cameras of the same make and model, there would not be any other solution than to not connect multiple of such cameras to same computer. But I didn't research this one any further, and the conclusion might be at least party incorrect.

To make sure there's not just a easily corrected mistake in how the cameras were configured, please attach also the output of the following commands. And if `v4l2-ctl` is not found on your system, install it first by running `sudo apt install v4l-utils`
```
v4l2-ctl --list-devices

ls -lr /dev/video*
```

Then there was this forum post: https://forums.raspberrypi.com/viewtopic.php?t=76558 but I'm not quite sure if the workaround of setting`input -1` (which in recent Motion versions would mean setting `video_params input=-1` would work with MotionEye, which (when the camera actually works) will set other video_params  (as can be seen from camera-1.txt you attached) and thus likely overwrites that.

dog2bert, 2024-03-22:
dog2bert@raspberrypi:~ $ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media3

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media0
        /dev/media1

rpivid (platform:rpivid):
        /dev/video19
        /dev/media2

UVC Camera (046d:0825) (usb-0000:01:00.0-1.2):
        /dev/video0
        /dev/video1
        /dev/media4

C270 HD WEBCAM (usb-0000:01:00.0-1.4):
        /dev/video2
        /dev/video3
        /dev/media5

dog2bert, 2024-03-22:
dog2bert@raspberrypi:~ $ ls -lr /dev/video*
crw-rw---- 1 root video 81, 13 Mar 19 18:17 /dev/video31
crw-rw---- 1 root video 81, 17 Mar 21 15:09 /dev/video3
crw-rw---- 1 root video 81,  7 Mar 19 18:17 /dev/video23
crw-rw---- 1 root video 81,  6 Mar 19 18:17 /dev/video22
crw-rw---- 1 root video 81,  5 Mar 19 18:17 /dev/video21
crw-rw---- 1 root video 81,  4 Mar 19 18:17 /dev/video20
crw-rw---- 1 root video 81, 16 Mar 21 15:09 /dev/video2
crw-rw---- 1 root video 81,  8 Mar 19 18:17 /dev/video19
crw-rw---- 1 root video 81, 12 Mar 19 18:17 /dev/video18
crw-rw---- 1 root video 81,  3 Mar 19 18:17 /dev/video16
crw-rw---- 1 root video 81,  2 Mar 19 18:17 /dev/video15
crw-rw---- 1 root video 81,  1 Mar 19 18:17 /dev/video14
crw-rw---- 1 root video 81,  0 Mar 19 18:17 /dev/video13
crw-rw---- 1 root video 81, 11 Mar 19 18:17 /dev/video12
crw-rw---- 1 root video 81, 10 Mar 19 18:17 /dev/video11
crw-rw---- 1 root video 81,  9 Mar 19 18:17 /dev/video10
crw-rw---- 1 root video 81, 15 Mar 21 15:09 /dev/video1
crw-rw---- 1 root video 81, 14 Mar 21 15:09 /dev/video0

zagrim, 2024-03-24:
It somehow seems to me, that the three cameras shown are actually the three devices under this one camera shown in v4l2-ctl output:
```
C270 HD WEBCAM (usb-0000:01:00.0-1.4):
/dev/video2
/dev/video3
/dev/media5
```
That also explains why only them works in ME, since if you were to check the capabilities of each device using `$ v4l2-ctl -d /dev/<device file name> --all`, likely only the first one (video2) would show "Video Capture" under "Device caps" (see https://unix.stackexchange.com/questions/541493/one-physical-device-mounts-two-dev-video-devices-which-will-cause-fswebcam-to and https://unix.stackexchange.com/questions/512759/multiple-dev-video-for-one-physical-device for some reference).

But I guess the big issue here is why none of the devices of that second camera aren't listed by ME, right?
You said you have two logitech C270 cameras plugged in, but `v4l2-ctl` showed one "UVC Camera" and one "C270 HD WEBCAM". The vendor ID + device ID pair "046d:0825" shown beside "UVC Camera" matches with Logitech Webcam C270, but I wonder why only the other has that shown and why the devices are not identified similarly.

What is the output of `lsusb` ?

dog2bert, 2024-03-25:
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 003: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

JamBalaya56562, 2026-11-12:
Sorry that this was left without an answer. Two cameras of the same model can end up with the same name under `/dev/v4l/by-id`, which motionEye uses to keep a camera's device stable; I could not check this without two identical cameras. As there was no reply since 2024, I'm closing this. If the second camera is still missing on the current version, please open a new issue with the output of `ls -l /dev/v4l/by-id /dev/v4l/by-path` and `v4l2-ctl --list-devices`.

closed 2026-11-12 by JamBalaya56562