motioneye-project/motioneye #2949: Motion detection with RTSP cameras - dont work normally

Aanistratenko1984, 2024-03-27:
Hi, community!
I work with video software like Hikvision, Dahua, Milestone without problems.
But need one for Linux so installed and test MotionEye for the first time.
Simple working software! 

But have one problem - setting up a motion detection is terrible!
Have many false records, moving objects are cutting out or dont record or record a pieces of its or with artefacts...
I tried many variant withou success (((
What mean "Noise Level" ? In other software i dont use same...
For example 
[image]

Please help me to select optiimal settings.
Thank you!

harakka, 2024-04-16:
Description for the parameters is here: https://motion-project.github.io/motion_config.html#OptDetail_Motion_Detection

Unfortunately there are no optimal parameters, they depend heavily on your situation. These are up to motion, not motioneye, and motion's model is pretty simple, it's essentially just looking at some treshold of changed pixels. Automatic treshold tuning and noise detection are usually best turned on.

JamBalaya56562, 2026-11-12:
harakka's answer covers it: the detection settings are Motion's (described at the link above), and the right values depend on the scene. Automatic Threshold Tuning and Automatic Noise Detection are a good start, and Show Frame Changes marks what Motion counts as change in the live view, which helps to set the threshold and the mask. Closing as answered.

closed 2026-11-12 by JamBalaya56562