motioneye-project/motioneye #2955: Slow preview and recording

cojsolorzano, 2024-03-31:
Greetings! I am using MotionEye and preview and recordings are slow, really slow, I set 30 FPS and it uses 2 or 3 FPS, so if a car passes by, it is never seen by MotionEye, I use another software in the same machine and it works well, 29 FPS. But I want to use MotionEye, I am using one USB Webcam. What settings should I consider to set? Thanks in advance.

ferociousdiablo, 2024-04-27:
try adding 

video_params v4l2_palette=8

to extra motion options in motioneye. It may be that its using YUV mode which can be much slower in my experience.

JamBalaya56562, 2026-11-15:
@cojsolorzano Many USB webcams reach their full frame rate only in MJPEG and drop to a few fps in YUYV at higher resolutions, which matches what you see. Did `video_params palette=8`, as @ferociousdiablo suggested, help? Until now motionEye dropped that option again when the camera settings were saved; from the next release it is kept (#3551). If it's still slow, please post the output of `v4l2-ctl -d /dev/video0 --list-formats-ext`.