motioneye-project/motioneye #2957: Install on Raspbian 12?

bomblord, 2024-04-05:
I have a new install headless Raspberry Pi OS setup on a Pi 4 "Debian GNU/Linux 12 (bookworm)"

When attempting to follow the following instructions  at https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian-Bullseye (which is the latest install instructions available) the system service fails to run. The error I've found in journalctl -u motioneye.service seems to point to the issue being a problem with not using pip2 based on an extremely old post here https://github.com/motioneye-project/motioneye/issues/1048

However Pip2/Python 2.7 is no longer supported or installable. I tried to upgrade using sudo pip install motioneye --break-system-packages --upgrade motioneye==0.42.1 but the service still will not start with the same error.

Since 0.43.1b1 specifically mentions python 3 support I also attempted that which changes the error but still does not resolve the issue.
`Apr 05 14:32:51 raspberrypi systemd[1]: Started motioneye.service - motionEye Server.
Apr 05 14:32:51 raspberrypi meyectl[795]: Traceback (most recent call last):
Apr 05 14:32:51 raspberrypi meyectl[795]:   File "/usr/local/bin/meyectl", line 5, in <module>
Apr 05 14:32:51 raspberrypi meyectl[795]:     from motioneye.meyectl import main
Apr 05 14:32:51 raspberrypi meyectl[795]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/meyectl.py", line 29, in <module>
Apr 05 14:32:51 raspberrypi meyectl[795]:     from motioneye import config, settings
Apr 05 14:32:51 raspberrypi meyectl[795]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/config.py", line 31, in <module>
Apr 05 14:32:51 raspberrypi meyectl[795]:     from motioneye import meyectl, motionctl, settings, tasks, uploadservices, utils
Apr 05 14:32:51 raspberrypi meyectl[795]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/motionctl.py", line 25, in <module>
Apr 05 14:32:51 raspberrypi meyectl[795]:     from tornado.httpclient import AsyncHTTPClient, HTTPRequest
Apr 05 14:32:51 raspberrypi meyectl[795]:   File "/usr/local/lib/python3.11/dist-packages/tornado/httpclient.py", line 50, in <module>
Apr 05 14:32:51 raspberrypi meyectl[795]:     from tornado import gen, httputil, stack_context
Apr 05 14:32:51 raspberrypi meyectl[795]:   File "/usr/local/lib/python3.11/dist-packages/tornado/httputil.py", line 107, in <module>
Apr 05 14:32:51 raspberrypi meyectl[795]:     class HTTPHeaders(collections.MutableMapping):
Apr 05 14:32:51 raspberrypi meyectl[795]:                       ^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 05 14:32:51 raspberrypi meyectl[795]: AttributeError: module 'collections' has no attribute 'MutableMapping'
Apr 05 14:32:51 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Apr 05 14:32:51 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.
`

Any assistance on this matter would be greatly appreciated. I could always go back to motioneyeos which still seems to work but that seems like a potential security issue.

zagrim, 2024-04-06:
The installation instructions in the Wiki all are for Python 2 based releases since the new Python 3 based on is not yet fully released, but can be installed using these instructions: https://github.com/motioneye-project/motioneye/blob/main/README.md
I don't remember if someone has used them already on Bookworm but at least they should work ok.

AntiSol, 2024-04-11:
I have installed on a couple of raspberry pis running bookworm using the instructions @zagrim linked to, and can confirm that they work :)

Bluscream, 2024-04-14:
> The installation instructions in the Wiki all are for Python 2 based releases since the new Python 3 based on is not yet fully released, but can be installed using these instructions: [...]

Sadly these instructions result in the folling error:

```
root@minopia:/tmp# /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 5, in <module>
    from motioneye.meyectl import main
  File "/usr/local/lib/python3.11/dist-packages/motioneye/meyectl.py", line 29, in <module>
    from motioneye import config, settings
  File "/usr/local/lib/python3.11/dist-packages/motioneye/config.py", line 31, in <module>
    from motioneye import meyectl, motionctl, settings, tasks, uploadservices, utils
  File "/usr/local/lib/python3.11/dist-packages/motioneye/motionctl.py", line 25, in <module>
    from tornado.httpclient import AsyncHTTPClient, HTTPRequest
  File "/usr/local/lib/python3.11/dist-packages/tornado/httpclient.py", line 50, in <module>
    from tornado import gen, httputil, stack_context
  File "/usr/local/lib/python3.11/dist-packages/tornado/httputil.py", line 107, in <module>
    class HTTPHeaders(collections.MutableMapping):
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'collections' has no attribute 'MutableMapping'
```

AntiSol, 2024-04-15:
@Bluscream this has to do with the change introduced in a recent python version which moved `MutableMapping` into `collections.abc`. Specifically it looks like you have a version of the `tornado` library which isn't compatible with python 3.11.

On my raspbian 12 machine I have tornado 6.4 (to check, do `pip freeze | grep tornado`) and it's been updated to be compatible with the change. I think you should be able to fix this with something like `pip install --upgrade tornado==6.4`.

Strange that it doesn't seem to have installed the correct version for you.

Bluscream, 2024-04-16:
Someone already pointed this out https://github.com/tornadoweb/tornado/issues/3372

Thank you tho. And this was on a fresh raspbian bookworm/motioneye install btw 🙄

Matthew1471, 2024-12-04:
For a fresh install these work too : https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345

JamBalaya56562, 2026-11-05:
motionEye runs on Python 3 since 0.43.1 (#1978), and the wiki page for Python 2 is gone. The Tornado error is gone too, as motionEye now requires a recent version. Checked with the current PyPI release on Debian 12 (Python 3.11) in a container: it installs and the server starts. `motioneye_init` and Raspberry Pi OS itself were not checked. For the current release, follow the [README on `main`](https://github.com/motioneye-project/motioneye/blob/main/README.md). Closing as completed; please reopen if those steps fail.

closed 2026-11-05 by JamBalaya56562