motioneye-project/motioneye #2959: Hide cameras from the "user" account?

bbb0, 2024-04-08:
Hi,

I would like to use Motioneye with two accounts - the admin account should see all cameras and settings as it is doing now. But the secondary user account now can see all the added cameras. Is there any way to make a flag on certain cameras just to hide them from the viewer account?

It helps me if someone points me to the rendering script, adding another if statement should be feasible... (if !camera.hidden then render)

xj25vm, 2025-06-24:
@MichaIng - yes - that is correct. One camera will only record motion footage at high resolution and high fps, the second camera will connect to the same physical camera, but the second substream - which is low resolution and low fps. This second camera will record continuously. This is in case there is something important going on, which might not be detected as motion - so there is at least some low fps/resolution footage of it. I don't mind if the admin sees all the cameras - as I only use the admin account for configuring cameras. But it is strange to explain to the users why there are 2 previews for every single camera - and it crowds out the interface

MichaIng, 2025-06-24:
It also causes unnecessary traffic and processing at the server. Probably easiest is to have camera preview switches per-user in admin settings. Might be also useful for the admin itself, if there are a lot of cameras.

xj25vm, 2025-06-25:
@MichaIng sounds like a good idea if it can be implemented

Marijn0, 2025-12-14:
If anyone is interested, I have tried implementing this in my local [branch](https://github.com/Marijn0/motioneye/tree/Camera-access-control) (with a little help from Copilot):
I have added a button for 'Admin only access' and it works pretty well for me. If anyone wants to test this:
```
sudo python3 -m pip install --force-reinstall --no-cache-dir --no-deps https://github.com/Marijn0/motioneye/archive/refs/heads/Camera-access-control.zip
```

MichaIng, 2026-02-01:
Done with #3245, many thanks @Marijn0!

closed 2026-02-01 by MichaIng