motioneye-project/motioneye #2961: Login > Remember me issue (0.43.1b1)

aigle1234, 2024-04-12:
Dear community,

It seems that the feature "Remember Me" is not working on version 0.43.1b1 (Debian 11 & Debian 10) => it always ask me to enter the username + password.
When I'm using version 0.42.1 (Debian 10), all is working fine (it doesn't ask me everytime the username + password).

[image]

twixi, 2024-04-24:
if you do not enter the login password, the authorization will work and you will see your cameras...

aigle1234, 2024-04-24:
Hello,
Yes by default I have no password and if I enter admin/empty, and then press enter, it's working.
The issue is that everytime I need to enter these credetials even if I have selected the "Remember Me" option.
On previous version, I don't have this behavior.

omumarvaishya005, 2024-08-07:
Thanks bro i tried alot to find username and it was fatal failure for me thanks to your answer know i know user name ans pass 
which is admin and nothing is there as such password (please guys don't put password as password you just enter admin and no need to put anything in password field).
thanksn bro and whom i helped thanks me latter