motioneye-project/motioneye #2964: New error when installing motioneye

Bluscream, 2024-04-14:
```
root@minopia:/tmp# pip install motioneye
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```
this didnt happen when i set up motioneye last time

Kvrnn, 2024-04-14:
It's in the [readme](https://github.com/motioneye-project/motioneye?tab=readme-ov-file#installation)

> On recent Debian (Bookworm ant later) and Ubuntu (Lunar and later) versions, the `libpython3.*-stdlib` package ships a file `/usr/lib/python3.*/EXTERNALLY-MANAGED`, which prevents the installation of Python modules outside of `venv` environments. motionEye however has a small number of dependencies with no strict version requirements and hence is very unlikely to break any Python package you might have installed via APT. To bypass this block, add `break-system-packages=true` to the `[global]` section of your `pip.conf`:
> ```bash
> grep -q '\[global\]' /etc/pip.conf 2> /dev/null || printf '%b' '[global]\n' | sudo tee -a /etc/pip.conf > /dev/null
> sudo sed -i '/^\[global\]/a\break-system-packages=true' /etc/pip.conf
> ```

Bluscream, 2024-04-14:
> It's in the [readme](https://github.com/motioneye-project/motioneye?tab=readme-ov-file#installation)

yeah yeah i have done all that, it wasn't related to it. What i actually had to do was run `python3 -m pip install tornado --upgrade` manually after installing motioneye. it upgraded from some ancient version to the latest and motioneye started right up afterwards

Bluscream, 2024-04-14:
> It's in the [readme](https://github.com/motioneye-project/motioneye?tab=readme-ov-file#installation)

You know there are about ~50 tutorials and readmes official or not and only one mentions this ...

xandark, 2024-07-25:
I have a suggestion: in the README.md, omit the part about grep'ing and chaning the pip.conf file and just add the **--break-system-packages** argument to pip like so:

`sudo python3 -m pip install --break-system-packages --pre motioneye`

Makes the installation easy again.

xandark, 2025-04-04:
I just reinstalled motioneye and this is even easier as I'm pulling the latest from git:

`sudo python3 -m pip install --break-system-packages git+https://github.com/motioneye-project/motioneye.git`

MichaIng, 2025-04-19:
In the end we need to rework install instructions and service files to use a venv.

I'll close this issue as answered regarding the `break-system-packages` option needed to install anything via `pip` outside of a venv on most modern distros.

closed 2025-04-19 by MichaIng