motioneye-project/motioneye #2969: RTSP and V4L2 cameras Grey screen issues

ejwjohn, 2024-04-24:
Have installed Motioneye 043.1b1. on a new image of Bullseye on Rpi 3B.

2 existing fully functioning networked cameras connected to Motioneye as RSTP using port 554 not getting camera image.

1 local USB camera connected as V4L2 same results as the RTSP cameras.

Ref the RTSP cameras tried TCP and UDP same results, also tried Sub and Main view options within the RTSP:// statement.

No Change.

MichaIng, 2024-04-24:
Please paste the motionEye server logs:
```sh
journalctl -u motioneye
```
Did you try it with only the USB camera and/or only 1 network camera, to rule out that it is a bandwidth or load issue? And does it work when you add the network camera as MJPEG camera (direct stream without passing through motion/motionEye), in case it does provide an MJPEG stream?

ejwjohn, 2024-04-24:
OK,
thanks for the info about the journal command, when I did that I noted that there were a lot of failures due to the camera failed authorization.

so I did a few tests and put in a known incorrect password in the camera setup stage and Motioneye accepted it??? this would appear to be a bug.

Now that I have resolved my password issue I was using an out-of-date version ..my mistake) I can now see the image for the RTSP cameras

However the directly connected USB camera still does not function, and when I execute the journalctl command you suggested  I cannot see any entries for today's date, as I wanted to see what is happening with the USB camera. I have no experience of using the journalctl command so could you give me some pointers please I have looked at the man page but cannot see anything of use. also is there a way to export this log file where is it stored exactly?

Thanks JohnW

ejwjohn, 2024-04-25:
Hello,
Ref my request for guidance on using journalctl I have found what I was looking for and managed to get the logs that impact the USB camera.

For some reason, motioneye thinks the camera I am using does not support capturing. which does not sound correct, I use this as a normal camera on my MAC and I have used it for Zoom etc. i have tried to configure it as MMAl but there system cannot find any compatible cameras?

ejwjohn, 2024-04-26:
One final comment is issue 2972 the same as my report here???

free-cup-holder-exe, 2026-10-11:
The "known incorrect password ... accepted" part is indeed a motionEye bug for cameras that use Basic authentication. The add camera check compared the authentication scheme as bytes with a string, so it never sent the credentials and always reported the camera as found. #3445 fixes that, so a wrong password gives "authentication failed" when adding the camera.

If a camera uses Digest authentication, motionEye still cannot check the password at that step, so a wrong one only shows up later as a grey picture and authorization errors in motion's log, as you found.

JamBalaya56562, 2026-11-15:
@ejwjohn About the USB camera: motion's "does not support capturing" usually means the selected device isn't the camera's capture node. Many USB webcams create two `/dev/video*` devices, and only one of them delivers pictures. Could you post the output of `v4l2-ctl --list-devices` and say which device you selected? Choosing the other one should fix it.