motioneye-project/motioneye #2979: Web Interface not working- log says "curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection Refused"

NbodySpecial, 2024-05-08:
First, thanks everyone for all the work you put into this project.  

---
OK, this one's got me stumped.  So it's probably a trivial thing right in front of me.

I have a pair of RasPI4's running Motioneye on Bullseye.  Or they were a few months ago when I last used them- now they're not working properly and I can't connect to the web UI . Same behavior on both. When I connect to <IP_addr>:8765 it says:

> This site can't be reached
> <IP_addr> refused to connect

Trying <IP_addr>/dashboard/  returns the same.

These are configured to write to a USB drive, so I pulled it and found it was taking 1 picture/second.  So at least something is working- but not the motion activation or the webUI.

Stopping/restarting the demons didn't have any effect.

Much testing later:
-Motion appears to be working.  
I can connect to <IP_addr>:7999 and it says
> `Motion 4.6.0 running [1] Camera`
I can connect to <IP_addr>:8081 and it shows the live picture from the camera.

I've stopped/restarted motioneye.  Changed the "Listen" line in /etc/motioneye/motioneye.conf to 0.0.0.0 and <IP_addr>, with no real effect (I see the changed IP in the log- otherwise the error is the same).

Netstat -tlupn | grep -e "motion|8765" returns the two motion ports as one would expect,  and python3 on 8765- with the pid on 8765 matching the pid of the motioneye process I find when I do "sudo systemctl status motioneye".

Increasing the log verbosity to debug didn't turn up anything I found useful.

Sudo systemctl status motion shows it as disabled (as it should be).

ifconfig shows the wifi on the RasPi is up, with the IP_addr being what I expect.

Connecting to the RasPi wifi and then doing tail -50 /var/log/daemon.log  I see the DHCPREQUEST and DHCPACK are there- my phone is successfully connecting to the RasPi's WiFi.

Not finding anything, I  broke out the hard wire ethernet and tried updating:
`sudo pip install motioneye --upgrade `
It said there was nothing to do- and no changes in function appeared.

I created a script to gather debugging info.  Note that before I ran it I'd manually started / stopped motion during testing, so it doesn't show as disabled, it shows as stopped.

```
# call with "sudo debug_server_issues.sh"
DEBUG_DIR=<your_debug_dir>
mkdir -p $DEBUG_DIR

echo "ifconfig results" > $DEBUG_DIR/ifconfig
echo "----------------------" >> $DEBUG_DIR/ifconfig
ifconfig >> $DEBUG_DIR/ifconfig

echo "systemctl status motion results" > $DEBUG_DIR/motion_status
echo "----------------------" >> $DEBUG_DIR/motion_status
systemctl status motion >> $DEBUG_DIR/motion_status

echo "systemctl status motioneye results" > $DEBUG_DIR/motioneye_status
echo "----------------------" >> $DEBUG_DIR/motioneye_status
systemctl status motioneye >> $DEBUG_DIR/motioneye_status

echo "netstat -tlupn | grep -E \"motion|8765|8081\" " > $DEBUG_DIR/netstat
echo "----------------------" >> $DEBUG_DIR/netstat
netstat -tlupn | grep -E "motion|8765|8081"  >> $DEBUG_DIR/netstat

cp /var/log/motioneye/motion.log $DEBUG_DIR
cp /etc/motioneye/motioneye.conf $DEBUG_DIR
cp /etc/motioneye/motion.conf $DEBUG_DIR
```

I've attached the results from this script.

Any suggestions?  

Thanks

[ifconfig.txt](https://github.com/motioneye-project/motioneye/files/15255797/ifconfig.txt)
[motion.conf.txt](https://github.com/motioneye-project/motioneye/files/15255798/motion.conf.txt)
[motion.txt.log](https://github.com/motioneye-project/motioneye/files/15255799/motion.txt.log)
[motion_status.txt](https://github.com/motioneye-project/motioneye/files/15255800/motion_status.txt)
[motioneye.conf.txt](https://github.com/motioneye-project/motioneye/files/15255801/motioneye.conf.txt)
[motioneye_status.txt](https://github.com/motioneye-project/motioneye/files/15255802/motioneye_status.txt)
[netstat.txt](https://github.com/motioneye-project/motioneye/files/15255803/netstat.txt)

NbodySpecial, 2024-05-09:
Hmm.. more digging online (without specifying motioneye or 8765) I found some similar errors on different ports.  (All the motioneye-specific pages I found weren't helpful)

Which lead me to try this command on the RasPi: 
`curl 127.0.0.1:8765`
And that returned a screenful of HTML.  Which supports the output of netstat showing python3 listening on that port.  

Now I'm wondering if the error in the logfile is just a timing issue (i.e. it's coming up, but isn't up yet when that curl command gets run in the log), and the real error lies elsewhere.

I don't have any firewall or proxy configured.
DHCP appears to be working (shown in original testing).

NbodySpecial, 2024-05-09:
I'd manually started/stopped motion, which is why that shows up.

We're thinking along the same lines - I've changed the  listen value in motioneye.conf multiple times.  (you can see those commented out).  Last night I was testing out more things and noticed that the :7999 connection had broken, but the :9081 was still working.  
Going from memory, the netstat at that time was:
tcp        0      0 0.0.0.0:8765          0.0.0.0:*               LISTEN      26721/python3       
tcp        0      0 127.0.0.1:7999            0.0.0.0:*               LISTEN      26730/motion        
tcp        0      0 0.0.0.0:9081            0.0.0.0:*               LISTEN      26730/motion  

So that would explain the loss of :7999, but still not explain :8765 not working.  

I'll confirm in a few hours when I get home.

NbodySpecial, 2024-05-10:
Got home and checked
-motioneye.conf had listen on 0.0.0.0
-motion.conf had webcontrol_localhost on

This gave a netstat result of:
tcp 0 0 0.0.0.0:8765 0.0.0.0:* LISTEN 26721/python3
tcp 0 0 127.0.0.1:7999 0.0.0.0:* LISTEN 26730/motion
tcp 0 0 0.0.0.0:9081 0.0.0.0:* LISTEN 26730/motion

With the ability to connect and see the video on 9081, and "refused to connect" on the others.

Changing motion.conf to "webcontrol_localhost off" and restarting  now gives a netstat of:
tcp 0 0 0.0.0.0:8765 0.0.0.0:* LISTEN 26721/python3
tcp 0 0 0.0.0.0:7999 0.0.0.0:* LISTEN 26730/motion
tcp 0 0 0.0.0.0:9081 0.0.0.0:* LISTEN 26730/motion

And: 
:9081 shows the video feed
:7999 gives "Motion 4.6.0 running [1] camera"
:8765 refuses to connect

Not sure if it's related, but with this setup I get the 

> "curl: (7) Failed to connect to 127.0.0.1 port 8765: Connection Refused"

message twice in /var/log/motioneye/motion.log.  With "webcontrol_localhost on" in motion.conf the message only appears once.

NbodySpecial, 2024-05-10:
To make it more interesting- and since nothing on the RasPi side was coming up wrong, I tried connecting from my phone using a different browser (had been using chrome).

<IP_addr>:8765   came up with the interface right away.  Tried again on chrome- no joy.

Went and got my laptop, connected, and opened Chrome.  UI came up just fine on :8765.

Restarted the phone.  Connected to the RasPi wifi, opened chrome.  UI came up fine on :8765.

ARRRRGGGHHH!!

Thanks for the help.  Hopefully at least this may help someone else in the future.

closed 2024-05-10 by NbodySpecial

MichaIng, 2024-05-10:
So you still see the "connection refused" in `motion.log` or has this been sorted? I actually remember seeing something similar, since the motionEye frontend seems to try to connect to the backend already (to get some settings, IIRC) before one has authenticated. Probably it is that one. While it is visual only, makes sense to track down this in code and skip this attempt until authentication has finished.

NbodySpecial, 2024-05-10:
Yes, I still see that - doubled in some cases as noted previously.

Thanks