motioneye-project/motioneye #2981: Raspberry 3b :Use composite video and network stream at the same time

Nudelnase, 2024-05-13:
Hello dear community,

maybe you can help me.

I have connected a USB camera to my Pi 3 B.

Now I want to use the composite output of the Pi (jack) to play the analog video signal on my old analog monitor.

At the same time, I would like to use the stream from the network to stream it to my browser or via VLC.

Question, is it possible to do this at the same time?

Why do I want to do this?

I don't want to have two cameras in one place, one analog and one IP camera.

I have already tried to install MotionEye, the network stream can be viewed on the computer. Unfortunately, I can't activate a picture on the composite output of the Raspberry 

Do you have any idea?

Best regards and thanks for your answers.

Translated with DeepL.com (free version)

MichaIng, 2024-05-13:
Good question. Which OS and version do you use?

On older RPi OS, `raspivid -t 0` seems to work, and on newer RPi OS `rpicam-vid -t 0`. I actually thought, that it would be for writing the stream to a video file or streaming it only, but when not defining an output, it seems to shows a "preview" window on the screen.

In case, you might need to install the package:
```sh
apt install rpicam-apps
```
Here are the official docs: https://www.raspberrypi.com/documentation/computers/camera_software.html

JamBalaya56562, 2026-11-12:
MichaIng's answer covers what can be done here: only one program can open the camera at a time, and motionEye itself does not show video on the composite output. To have both, one program, e.g. `rpicam-vid`, would have to show the picture on the screen and serve a network stream, which can then be added to motionEye as a network camera. Closing as answered.

closed 2026-11-12 by JamBalaya56562