motioneye-project/motioneye #2982: Motioneye on DietPi

UKnau22, 2024-05-14:
I installed MotionEye on a RaspberryPi 4 with DietPi, using the dietpi-software command.

I spent a whole day trying to figure out how to add my new Reolink camera and get an rtsp stream. I also tried adding a camera that I know is working with MotionEye. Turns out there's nothing wrong with the camera or my settings. I was able to add the Reolink camera without any issues in my MotionEye installation on HomeAssistant, which is where I also run the camera I tried adding to the DietPi setup.

So what's happening is that I can add the camera rtsp stream, but the camera view actually is just grey, nothing visible.

I reinstalled MotionEye to no avail

Anybody have any experience with that or idea what might be wrong? Some dependencies issues?

MichaIng, 2024-05-14:
The Home Assistant integration uses the old Python 2 based motionEye 0.42.1 code, which differs quite significantly from what is installed on DietPi or when you follow these instructions: https://github.com/motioneye-project/motioneye?tab=readme-ov-file#installation

Also, the underlying `motion` version, is likely different, as well as `v4l-ctl` and `ffmpeg`, respectively the used FFmpeg libraries. Means, there is no point to compare both setups, but instead we need to focus on why it does not work with the bare metal motionEye Python 3 module.

Just to be sure, it is a USB camera, added as "V4L2" camera to motionEye? Does motionEye throw any related errors?
```sh
journalctl -u motioneye
```

_Btw, as a general recommendation: When you have questions or face issues with a distribution's implementation of some software, report/ask it to the distribution first. Upstream developer might not know which distribution implemented it how, which version they use, whether they apply patches etc. Similarly, I have no detailed idea how HA implemented their motionEye integration. The distribution maintainer(s) can then redirect it upstream, if it turns out to be an upstream issue. However, in this case, I know it is an upstream issue, since I am the DietPi maintainer and know that we implemented motionEye just like it is advertised on the readme linked above 😉._

UKnau22, 2024-05-14:
I did read a few posts on the DietPi forum and concluded that this is the better place to ask...

Yes, I added it as a network camera with rtsp://URL

Following errors are thrown, the first one several times:

May 14 11:12:20 MotionEye meyectl[669]:    ERROR: ffmpeg: could find version: Command '['/usr/bin/ffmpeg', '-version']' returned non-zero exit status 127.

May 14 11:12:20 MotionEye meyectl[669]:    ERROR: 500 POST /config/add/?_=1715710340275&_username=*****&_signature=*********************************************************** (192.168.1.147) 518.23ms
May 14 11:12:21 MotionEye meyectl[669]:    ERROR: motion not running, starting it
May 14 11:12:22 MotionEye meyectl[669]:    ERROR: failed to start motion: motion failed to start

May 14 11:12:32 MotionEye meyectl[669]:    ERROR: failed to start motion: motion failed to start
May 14 11:12:32 MotionEye meyectl[669]: Traceback (most recent call last):
May 14 11:12:32 MotionEye meyectl[669]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/server.py", line 374, in checker
May 14 11:12:32 MotionEye meyectl[669]:     motionctl.start()
May 14 11:12:32 MotionEye meyectl[669]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/motionctl.py", line 127, in start
May 14 11:12:32 MotionEye meyectl[669]:     raise Exception('motion failed to start')
May 14 11:12:32 MotionEye meyectl[669]: Exception: motion failed to start

UKnau22, 2024-05-14:
Here's the output of that....

root@MotionEye:~# which ffmpeg
/usr/bin/ffmpeg
root@MotionEye:~# ls -l /usr/bin/ffmpeg
-rwxr-xr-x 1 root root 330136 Feb 28 07:38 /usr/bin/ffmpeg
root@MotionEye:~# dpkg -l ffmpeg
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version                Architecture Description
+++-==============-======================-============-================================================================
ii  ffmpeg         8:5.1.4-0+rpt3+deb12u1 arm64        Tools for transcoding, streaming and playing of multimedia files

MichaIng, 2024-05-14:
That looks all good. And the command motionEye failed to execute, does it work?
```sh
/usr/bin/ffmpeg -version
sudo -u motion /usr/bin/ffmpeg -version
```

UKnau22, 2024-05-14:
root@MotionEye:~# motionEye
-bash: motionEye: command not found

This doesn't seem right... I'm sorry, not a pro with command line....

MichaIng, 2024-05-14:
I did not mean the command "motionEye", but the command which motionEye failed to run 😄. Please execute the commands from the code box instead.

UKnau22, 2024-05-14:
Hmmm.... I tried that all kinds of ways, below is the output. Sorry if I'm being stupid...

root@MotionEye:~# /usr/bin/ffmpeg -version
/usr/bin/ffmpeg: symbol lookup error: /lib/aarch64-linux-gnu/libgio-2.0.so.0: undefined symbol: g^list_foreach

root@MotionEye:~# -u motion /usr/bin/ffmpeg -version
-bash: -u: command not found

root@MotionEye:~# /usr/bin/ffmpeg -version
sudo -u motion /usr/bin/ffmpeg -version
/usr/bin/ffmpeg: symbol lookup error: /lib/aarch64-linux-gnu/libgio-2.0.so.0: undefined symbol: g^list_foreach
/usr/bin/ffmpeg: symbol lookup error: /lib/aarch64-linux-gnu/libgio-2.0.so.0: undefined symbol: g^list_foreach

MichaIng, 2024-05-14:
There we have the issue. This library is from the `libglib2.0-0` package. `g^list_foreach` looks like the file is corrupted. It should be `g_list_foreach`. Please try to reinstall that package:
```sh
apt install --reinstall libglib2.0-0
systemctl restart motioneye
```

UKnau22, 2024-05-14:
That absolutely did the trick! Thank you so much! I spent all day yesterday trying to figure out what I'm doing wrong....

closed 2024-05-14 by UKnau22

MichaIng, 2024-05-14:
Great. That was not an easy one, indeed. Reading the output, I am about to enhance the logging a little: https://github.com/motioneye-project/motioneye/pull/2983/files

Though, the most important output about the library error is still missing. The `CalledProcessError` returned only the exit code:
> ```
> Command '['/usr/bin/ffmpeg', '-version']' returned non-zero exit status 127.
> ```

while I would like to (additionally) see the actual error there:
> ```
> /usr/bin/ffmpeg: symbol lookup error: /lib/aarch64-linux-gnu/libgio-2.0.so.0: undefined symbol: g^list_foreach
> ```

And I even interpreted the exit code 127 wrong in this case, as the shell returns it when an executable was not found. In this case, however, the command was found and executed, a shared library loaded, and only within this library, a symbol was found undefined (due to false character/corruption). No idea why this results in 127 as well 😄.

Probably the command's STDERR was stored in that `output` variable as well, so that we can just append it to the error message.

UKnau22, 2024-05-14:
Was any of that English?! 🤣🤣🤣

Well you did figure it out and are obviously much more well versed in this

Thank you!

-------- Original Message --------