motioneye-project/motioneye #2984: Lovelace cards stay grey, however the web-interface of motioneye shows the cams.

An0n1muS1994, 2024-05-16:
[image]
[image]

Also i see this error:
[image]

An0n1muS1994, 2024-05-16:
[image]

zoidy, 2024-05-17:
You'll probably have better luck asking your question on the Home Assistant GitHub repo...

JamBalaya56562, 2026-11-12:
As the cameras show in motionEye's own web interface, the grey cards come from the Home Assistant side, as zoidy said; the Home Assistant integration is the place to ask. If the card uses a motionEye URL, the Snapshot URL and Streaming URL under Video Streaming → Useful URLs can be opened in a browser to check that motionEye answers them. Closing here.

closed 2026-11-12 by JamBalaya56562