motioneye-project/motioneye #2988: Camera not being detected, no matter what

letkan, 2024-05-27:
I've tried Motioneye on Bookworm and Bullseye and while I get no errors while installing Motioneye and I can access it from the browser, the camera never works. I've looked everywhere for solutions but none worked. Funny enough, MotioneyeOS works perfectly. I just wanted to run it on a recent version of Debian. I've used the instructions from _https://github.com/motioneye-project/motioneye/blob/main/README.md_. I've also read a very long thread in Issues about the camera not working but the conclusion seemed to be that the issue has been fixed. Here's some more info of my seup or Raspberry pi Zero 2 w:
```
motionEye Version 	0.43.1b1
Motion Version 	4.6.0
OS Version 	Debian 11
```
libcamera-hello:

```
[0:09:11.886165531] [725]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:09:11.964724466] [726]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media1 and ISP device /dev/media0
[0:09:11.964889827] [726]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:09:11.967075670] [725]  INFO Camera camera.cpp:1033 configuring streams: (0) 1296x972-YUV420
[0:09:11.968575225] [726]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
```

letkan, 2024-05-29:
FYI, Motioneye same version works on DietPi.

Marijn0, 2024-06-02:
Have you already tried enable legacy camera stack:
https://projects.raspberrypi.org/en/projects/getting-started-with-picamera
(as far as I know this only works in Bullseye)

And I had to load the bcm2835-v4l2 kernel module:
https://github.com/motioneye-project/motioneye/wiki/Troubleshooting

letkan, 2024-06-02:
Legacy camera only works in Bullseye indeed, as I mentioned earlier, I switched to Dietpi Bookworm and the app works without any changes. I'm rather new to this platform and a lot of info available is outdated, for the time being I'll stick with "out-of-the-box" solutions. Thanks for replying, anyway.

letkan, 2024-06-10:
Everything finally came into place. More reading, more trial and error.

MichaIng, 2024-06-10:
Legacy camera support has been removed from the new RPi firmware/kernel stack, RPi OS Bookworm ships with. Since it requires some other reworks, DietPi did not switch to this new stack with their Bookworm images yet, hence it does still work.

However, this is indeed not future prove. When switching to the new kernel stack and hence modern libcamera API, there seems to be no other way for motionEye to support it, but using the `libcamerify` wrapper. See some issues here, which contain info about how to set this up: https://github.com/motioneye-project/motioneye/issues?q=is%3Aissue+is%3Aopen+libcamerify

When I find time, I'll implement it as option, and finally probably pre-install on RPi with `motioneye_init`. A new camera type for RPi libcamera modules will then use libcamerify as motion wrapper automatically.

closed 2024-06-10 by MichaIng