motioneye-project/motioneye #2990: Cannot add camera with 0.43.1b1

developerfromjokela, 2024-05-28:
[image]

motionEye Version 	0.42.1
Motion Version 	4.3.2
OS Version 	Ubuntu 22.04

zagrim, 2024-05-30:
motionEye Version 0.42.1 is Python2 only, it won't work with Python 3.12 which you seem to have. Please install the prerelease of 0.43 using [these instructions](https://github.com/motioneye-project/motioneye?tab=readme-ov-file#installation).

zagrim, 2024-05-30:
However as I now notice the title of this issue mentions 0.43.1b1: which one is correct?

developerfromjokela, 2024-05-30:
I start the script with python3, so why is it picking up py2 libs?

developerfromjokela, 2024-05-30:
> However as I now notice the title of this issue mentions 0.43.1b1: which one is correct?

?

developerfromjokela, 2024-05-30:
Ignore the versio 0.42, it's 0.43

developerfromjokela, 2024-05-30:
This error seems to occur when network camera is unavailable, so poor error handling here. Working cameras don't throw this error

zagrim, 2024-05-30:
The version confusion came from your problem description which specifically said "motionEye Version 0.42.1". But ok, if you are really running 0.43.x then fine, let's move on.

Having the full stack trace would likely make it easier to figure out the reason things go wrong (=why the callback passed to tornado lacks a positional argument), unless someone with more experience on dealing with Tornado can guess the cause from that snippet you posted.

Please also post the output of this command in case there's some sort of version incompatibility:
```
pip freeze | grep tornado
```

developerfromjokela, 2024-05-30:
There are no more stack traces in logs.

I'll send tornado version later.

free-cup-holder-exe, 2026-10-11:
You were right that this is poor error handling in motionEye, not your setup. When an RTSP camera does not answer within 10 s, motionEye's timeout calls `on_connect()` without the connection it needs, hence `on_connect() missing 1 required positional argument: 'f'`. The add camera dialog then waits until the operating system gives up the connection (over 2 minutes in my test) instead of reporting "timeout connecting to rtsp netcam" after 10 s. #3444 fixes it.

To answer the question above about the missing argument: the timeout is registered as `functools.partial(on_connect, _timeout=True)`, so it calls `on_connect(_timeout=True)` without the future that `on_connect(f, _timeout=False)` requires.

So the error itself means motionEye could not reach the camera at that address and port. If the camera works elsewhere, check the host, port and path in the RTSP URL.

developerfromjokela, 2026-10-11:
> You were right that this is poor error handling in motionEye, not your setup. When an RTSP camera does not answer within 10 s, motionEye's timeout calls `on_connect()` without the connection it [...]

I've moved on meye long time ago :) but nice this is getting some love it deserves.

closed 2026-10-11 by MichaIng