motioneye-project/motioneye #3005: Arch aarch64 systemd fix

letkan, 2024-06-10:
So I'm running MotionEye on [Arch aarch64](https://github.com/fwcd/archlinuxarm-images/releases/download/2024-06-09/archlinux-rpi-aarch64-fwcd-2024-06-09.img.zip) and it's been installed from git as per instructions. All is well except 
 `/etc/systemd/system/motioneye.service` which points to `/usr/local/bin/meyectl` instead of `/usr/bin/meyectl` where it's been actually installed.
Might be just a distribution specific issue, I don't know.

MichaIng, 2024-06-10:
Arch Linux is installing `pip` packages to `/usr` instead of `/usr/local`? All distros I use consequently use `/usr/local`, and keep `/usr` for their distribution package manager exclusively.

We can do `which meyectl` or `command -v meyectl` to detect the effective executable location, so not a big issue. I am just surprised.

letkan, 2024-06-10:
> Arch Linux is installing `pip` packages to `/usr` instead of `/usr/local`? All distros I use consequently use `/usr/local`, and keep `/usr` for their distribution package manager exclusively.

Yes, `/usr/`
> 
> We can do `which meyectl` or `command -v meyectl` to detect the effective executable location, so not a big issue. I am just surprised.

MichaIng, 2024-06-17:
PR up: #3011

Does it work when you run `motioneye_init` with this branch?
```sh
sudo python3 -m pip install --upgrade --force-reinstall --no-deps 'https://github.com/motioneye-project/motioneye/archive/arch.tar.gz'
sudo motioneye_init
```

MichaIng, 2024-10-29:
Fixed for next release/beta iteration.

closed 2024-10-29 by MichaIng