motioneye-project/motioneye #3006: Feature Request - User permissions

NicKerrr, 2024-06-11:
Hello! It would be useful to have the ability to set user viewing permissions for a specific set of cameras, allowing for more granular control over who can access cameras.

MichaIng, 2024-06-11:
At first, there would need to be a multi-user system, which currently does not exist: There is only the admin user for admin tasks, and a single possible surveillance user, which can view all camera streams.

So I'll add your request here, and close this: #541

closed 2024-06-11 by MichaIng

MichaIng, 2026-02-01:
A start was done with #3245, many thanks @Marijn0!

As much granular as it gets with only one non-admin user 😅. A multi-user management is still a dedicated open request as posted above.