motioneye-project/motioneye #3007: Pull request for the Wiki: Updated the "Install In Docker" page

mdPlusPlus, 2024-06-13:
I've cloned the Wiki according to https://gist.github.com/larrybotha/10650410.  
Unfortunately GitHub doesn't let you issues pull requests for wiki pages.  
Please check the link above to see how to merge the changes (if deemed adequate).
(The resulting repo links are https://github.com/mdPlusPlus/motioneye.wiki.git or git@github.com:mdPlusPlus/motioneye.wiki.git for easy access)

Changes:  
- replaced all references of the terribly outdated Docker Hub repository with the GitHub Container Registry one
- replaced all mentions of https://github.com/ccrisan/motioneye with https://github.com/motioneye-project/motioneye
- removed the instructions for building and running on the Raspberry Pi as the GHCR images work differently (should probably be readded by someone who knows more about the Pi building process than me)
- improved syntax and aesthetics

Reasoning:  
I think a lot of people refrain from installing MotionEye because of the Python2 dependency. A nice way to work around this is Docker. However, until now the images referenced in the wiki were terribly outdated (3+ years) and do not receive updates anymore. The devs have long switched to GHCR as replacement.  
Now people who want to try the docker images will be directed to the correct images that will keep getting updated.

mdPlusPlus, 2024-06-13:
Apparently there isn't even a Python 2 dependency anymore. In this case there are propably a lot more lines in the wiki that should be changed.

MichaIng, 2024-06-15:
Looks mostly fine. However, I would completely remove method 3 (the Python 2/motionEye v0.42.1 build). We cannot support this ancient versions with its very own problems anymore, and we should not document how to install a dead-end environment. There is the `python2` branch, so if someone for whatever reason really needs it, it can be figured out from there.

The timezone list is fine, but it is somehow weird for have the PHP documentation linked. Suggested replacement: https://en.wikipedia.org/wiki/List_of_tz_database_time_zones

Generally, it would be great to update and align the README from the Docker dir in the same turn: https://github.com/motioneye-project/motioneye/blob/dev/docker/README.md
Not great to have duplicate instructions, but as long as they can be aligned via copy&paste, it is okay for me, so there are instructions when cloning the repo, as well as in the wiki, where people commonly check for when visiting the GitHub page.

MarcelCoding, 2024-06-16:
The documentation itself looks great. I would suggest to move the wiki to a docs folder inside the repo which would improve the entrance to contributions and also link the documentation to the appropriate version. Additionally I would replace the wiki just with a link to the markdown files. As a last step the pull request template would be updated to inform contributors that the documentation should be updated in the same pull request as features are added. This could improve the documentation quality and correctness.

MichaIng, 2024-06-16:
That is a very good idea. The wiki could still contain some meta information, like the available branches. The benefit of merging all docs into an own directory is that we could generate a versioned MkDocs site with this, using e.g. GitHub Pages.

closed 2026-10-29 by Marijn0