motioneye-project/motioneye #3015: YUV420P issue with decoder = vaapi

mdPlusPlus, 2024-06-23:
I'm using following lines in my camera.conf to use hardware decoding (VA-API):

```
...
netcam_url rtsp://mycamera.lan:12345/stream
netcam_params decoder = vaapi,keepalive = off,tolerant_check = on,rtsp_transport = tcp
...
```

If I check my motion.log, I find the following hint:

```
netcam_rtsp_ntc: ******************************************************
netcam_rtsp_ntc: The image sent is being 
netcam_rtsp_ntc: trancoded to YUV420P.  If possible change netcam 
netcam_rtsp_ntc: picture format to YUV420P to possibly lower CPU usage.
netcam_rtsp_ntc: ******************************************************
```

And I notice a small CPU utilization increase, which makes me think the format transformation is done on CPU, before being sent to GPU.

However, when I check the actual pixel format of the stream it _is_ YUV420P:

```
$ ffprobe -hide_banner rtsp://mycamera.lan:12345/stream

[rtsp @ 0x55e2cb8ad700] max delay reached. need to consume packet
[rtsp @ 0x55e2cb8ad700] RTP: missed 1 packets
Input #0, rtsp, from 'rtsp://mycamera.lan:12345/stream':
  Metadata:
    title           :  
  Duration: N/A, start: 0.162811, bitrate: N/A
  Stream #0:0: Video: h264 (High), yuv420p(progressive), 1920x1080, 30 fps, 30 tbr, 90k tbn
```

Is this somehow user error or is there something wrong with the detection of the used pixel format?  
I see there is an `input_format` option for `netcam_params`, but from the documentation it does not seem related.

AleXburnA, 2024-12-30:
I have exactly the same question. I managed to get vaapi working but I see the same warning in the logs and I also see a slightly higher CPU usage, while the GPU is used as well. (Checked with intel_gpu_top)

I also checked with ffprobe and can confirm that my source camera stream is yuv420p

JamBalaya56562, 2026-11-12:
This notice comes from Motion, not motionEye: Motion shows it whenever a decoded frame is not YUV420P. With a hardware decoder, Motion copies each frame back from the GPU, and that copy comes back in the GPU's format, typically NV12, even when the stream itself is yuv420p (Motion's cuda and drm paths ask for NV12 explicitly). So with `decoder = vaapi` the notice and the conversion are expected and cannot be changed from motionEye. The config asked about above is the one in the first post: `decoder = vaapi` in `netcam_params`. Closing as answered.

closed 2026-11-12 by JamBalaya56562