motioneye-project/motioneye #3016: "Device or resource busy"

rcho5840, 2024-06-25:
I have a raspberry pi 4 set up with motioneye 0.43. Everything ran smoothly with the install and accessing the motioneye interface. The cameras also registered when I plugged them in via usb. However, I only see a gray screen with the cameras. Checking the motioneye log, this is what I see. I didn't want to paste the whole code but it repeats those last 5 lines for everytime I tried to set it up.

/[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.6.0 Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1:Camera1] [ERR] [VID] v4l2_input_select: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera

Also, weirdly, I see multiple cameras with my usb camera's name when I'm adding a camera even though I only have one camera plugged in right now.

[image]

I think it's possible that something else is using my camera and therefore the "device or resource is busy." fyi, the camera light is also on.

Marijn0, 2024-06-25:
Try:
sudo systemctl stop motion
sudo systemctl disable motion

MichaIng, 2024-06-27:
This should have happened automatically when running `motioneye_init` after: #2980

You installed it right before your report?

Marijn0, 2024-06-27:
This is probably the cause:
https://github.com/motioneye-project/motioneye/issues/2373#issuecomment-2138222114

MichaIng, 2024-06-28:
Oh right, I forgot about that. @ccrisan could you check your PyPI account? If you are short in time to maintain it, probably it is possible to add another project maintainer on PyPI or hand it to someone else? Would be great to keep using the original package name 🙂.

ccrisan, 2024-06-28:
> Oh right, I forgot about that. @ccrisan could you check your PyPI account? If you are short in time to maintain it, probably it is possible to add another project maintainer on PyPI or hand it to someone else? Would be great to keep using the original package name 🙂.

I just "verified" my email address as well as enabled 2FA on my PyPI account. Can we retry?

MichaIng, 2024-06-30:
Ah great, that worked, many thanks ❤️: https://pypi.org/project/motioneye/#history

closed 2024-06-30 by MichaIng