motioneye-project/motioneye #3021: Still Images: cams take pictures 24/7

SuMaPa, 2024-06-30:
hi

I use MotionEye Version 0.43.1b1, Motion Version 4.6.0, OS Version Debian 12 on Raspberry Pi5.
python version: 3.11.2
I have set in my settings:
Still Images: Motion Triggered [(One Picture)]
Working Schedule: Mo-Su 0600 - 2200, Outside Working Schedule
Text Overlay, Video Streaming, Movies & Motion Notifications are deactivated.

But the cams take pictures 24/7. 
I have no idea what could be causing this.

best regards

[image]
[image]
[image]
[image]
[image]
[image]
[image]

MichaIng, 2024-06-30:
This has been fixed with motionEye 0.43.1b2 already, but we currently cannot publish to PyPI, because the original maintainers email needs to be refreshed. Please update it like this for now:
```sh
sudo python3 -m pip install --upgrade --force-reinstall --no-deps 'https://github.com/motioneye-project/motioneye/archive/0.43.1b2.tar.gz'
```

SuMaPa, 2024-06-30:
Many thanks Micha!!
I took your first option.
I would like to test it by tomorrow evening.
Then I will set it to solved.

SuMaPa, 2024-07-02:
hi
Thank you very much, works very well!!