motioneye-project/motioneye #3022: Unable to open Video Camera

martbrad11, 2024-06-30:
Pi4B Bullseye
Arducam 'Night Day' (OV5647)
I can capture jpg with libcamera-jpeg and I can stream raw h264 to a VLC player so I know my camera is connected and working.
However when I try to add it in MotionEye I get this in the logs  vid_start: V4L2 device failed to open
I'm trying to add it as a Local V4L2 camera. If I try as MMAL I just get 'An error has occured try refreshing@
Can someone point me in the direction to get my camera working in MotionEye?

MichaIng, 2024-06-30:
`motion` does not support libcamera. Hence you need to either switch to legacy camera support, and use MMAL driver, or use `libcamerify` to start motionEye with libcamera=>V4L2 API. See some examples here: https://github.com/motioneye-project/motioneye/issues?q=is%3Aissue+is%3Aopen+libcamerify

JamBalaya56562, 2026-11-13:
As @MichaIng said, Motion cannot use libcamera directly. motionEye 0.45.0 starts it through libcamera's V4L2 layer itself when `libcamera_v4l2` is set (#3546; `sudo /opt/motioneye/bin/motioneye_init --libcamera` sets it up), and the camera can then be added as a local V4L2 camera. Until that release, the `libcamerify` steps in the wiki's [Troubleshooting](https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#libcamera-camera-stack) page do the same. Closing.

closed 2026-11-13 by JamBalaya56562