motioneye-project/motioneye #3023: Cameras not recording

rcho5840, 2024-07-01:
I have tried to use motion detection recording and although I see the red outline, there are no videos that are saved. Now, keep in mind that I do not have a flash drive in. My motioneye is setup on a raspberry pi with a microsd card. I was just checking if my video was actually saving before I used my flashdrive. Would this issue be fixed if I used a flashdrive? It's coming in the mail.

[image]
[image]

The second issue I noticed was that even with the continuous recording, there is no red outline. I believe this means that it's just not recording in general.

[image]

Martypose, 2024-07-08:
change movies to this

[image]

change the codec, that was my problem, default didnt work

rcho5840, 2024-07-08:
@Martypose yup one of the issues was the codec. It seems for my camera, it only wants to use mpeg however, so there's that.

zagrim, 2024-07-22:
Recording mode "Continuous recording" should mean just what it says, and it actually does the same as adding `emulate_motion on` in Extra Motion Options. And it should show the red outline.

Could you @rcho5840 please post the logs (`sudo journalctl -u motion`) so we can see if there are some underlying issues with your config?

Please also clarify this part:
> Now, keep in mind that I do not have a flash drive in. My motioneye is setup on a raspberry pi with a microsd card. I also have a flashdrive with a custom path set to the flash drive storage

Do you mean that you have not connected the flash drive, but you still have the flash drive path set in "File storage" > "Root directory"? That would obviously not work, but maybe I read your words wrong :smiley:

zagrim, 2024-07-22:
I also noticed that you'd opened #3025 and I wonder if that was resolved and if these two issues are actually related?

Matthew1471, 2024-12-04:
I came here to raise that OMX doesn't work too.. I got errors from motion saying it couldn't set the codec but ffmpeg does seem to have OMX compiled in.

ToeKneeFan, 2026-02-26:
I know this is old/closed, but I thought I would share in case it helps anyone else. (And my problem was the same as this issue.)

At least for me, the standard ffmpeg was not working properly with `h264_v4l2m2m` out of the box and was issuing errors. Switching to H.264 fixes this, but that forces software encoding, resulting in poor performance. However, manually compiling jc-kynesim's [ffmpeg fork](https://github.com/jc-kynesim/rpi-ffmpeg/) with patches/revisions for Raspberry PI and using that instead worked.

This is actually not unlike OMX back in the day (when that was previously supported in the legacy APIs).  Users needed to patch and manually compile it ([1](https://github.com/motioneye-project/motioneye/issues/930#issuecomment-522848452)). For that reason, a nonstandard patched version of ffmpeg was incorporated into motioneyeos.

ToeKneeFan, 2026-03-06:
For sake of clarification, on the old Raspbian Buster with motionEye v0.42.1, it is set up as
```
mmalcam_name vc.ril.camera
```

On the new Diet Pi Trixie with motionEye dev, it is set up as
```
video_device /dev/video0
``` 

> I just recognized that motion 4.7.0 removed the `mmalcam_name` option, so we should remove the MMAL handling entirely as well.

Ah, that must be why the newer instance needs the V4L2 MMAL service instead of adding directly through `mmalcam_name`.