motioneye-project/motioneye #3024: Flashdrive not showing up under stoarge device

rcho5840, 2024-07-01:
Hello, I have a flashdrive connected to my raspberry pi. I know it is recognized by the pi, because I see the name of the flashdrive when I do "df -h"

However, the flashdrive is not being recognized by motioneye. The only available storage device is custom path:

[image]

rcho5840, 2024-07-02:
Anyone have any feedback?

MichaIng, 2024-07-02:
I can replicate it. Just attached a test drive and mounted its filesystem. It appears in `fdisk` output as well as in `/dev/disk/by-id/` (which are both used by motionEye to check for available disks) but only "Custom Path" can be selected.

I'll have a closer look into this tomorrow. Busy today with my other project.

rcho5840, 2024-07-03:
I was able to mount the drive as well and make it permanent. I am just using custom storage now, but I'm having a new issue. mycreate_path is saying that permission is denied to even upload the media to the flash drive.

MichaIng, 2024-07-03:
You need to grant the `motioneye` user or group write access to that directory.

zagrim, 2024-07-22:
@rcho5840 were you able to resolve this?

MichaIng, 2026-10-12:
R/O mounts are now listed, but greyed out (not selectable) with a label indicating that they are not writable.

closed 2026-10-12 by MichaIng