motioneye-project/motioneye #3025: Permissions for video

rcho5840, 2024-07-03:
Does anyone know how to solve this issue? I am in 0.42, using a raspberry pi 4 with 2 webcams and a flashdrive attached.

[image]
[image]

I am having a lot of trouble figuring this out. I have written my target_dir in motion.conf, ran the commands listed below, mounted my flashdrive to a permanent directory. I don't know what I can do at this point.

[image]
[image]

MichaIng, 2024-07-03:
You need to run those commands with the mount point of the flash drive as argument. Should be obvious, but from the order you write things, it sounds like you probably ran the commands first (with `/home/pi/Monitor` literally?) and mounted the flash drive afterwards.

rcho5840, 2024-07-03:
No. I put in /media/anomaly, which is the directory that I wish to store my data. Even with those changes, and the rest of the changes above, I am still getting denied permissions

MichaIng, 2024-07-03:
Ah, I see you use 0.42.1, please switch to 0.43.1 and try again: https://github.com/motioneye-project/motioneye?tab=readme-ov-file#installation
And verify that the motion user or group has at least execute permissions on any parent directory:
```sh
ls -dl / /media /media/anomaly
```