motioneye-project/motioneye #3026: how setup android without port forwarding on router (e.g. ngrok?) and connect it to MotionEye cloud

ttodua, 2024-07-06:
it is possible that Android phone can be used as a camera (using IP Webcam) with ngrok? any tutorial out there?

MichaIng, 2024-07-06:
Best is probably a VPN, like WireGuard, which is quite easy to setup on the Linux system (which runs motionEye), and has a simple Android app.

Then you can use an app like DroidCam to stream the phone's camera, which can then be added as network camera to motionEye, when connected via VPN.

ttodua, 2024-07-07:
@MichaIng many thanks for response, I will try it. also, WireGuard was new to me !

JamBalaya56562, 2026-11-05:
As MichaIng wrote above, a VPN such as WireGuard does this without port forwarding: connect the phone to the system running motionEye, stream the camera with an app like DroidCam, and add that stream as a network camera. Closing as answered; please reopen if you get stuck on the motionEye side.

closed 2026-11-05 by JamBalaya56562