motioneye-project/motioneye #3027: How change directory in fille storage

mariogro, 2024-07-08:
Hi everyone, I'm a newbie to Motioneye. 

I installed Motioneye in a Docker container on Debian 12. 
I would like the video storage to be an ext4 USB disk which Debian sees as /media/usbdisk.
I log in Debian  as user, but I have RW permissions on the external disk and I am in the sudoers group. I log in Motioneye as Admin
In the Camera File Storage section on Motioneye I wrote as path /media/usbdisk, but Motioneye writes the videos to /var/lib/docker/overlay2/b97c14acacefff5556cbb6e4542b770f6255cb3d6755e4806a745e18c40da959/diff/media/usbdisk.

How do I tell motioneye the right way? Thank you and sorry for my english

shaynethiessen, 2024-07-14:
You should mount /media/usbdisk to your docker container. Depending on your configuration, that can be done using -v command or by adding it to your docker-compose file.

There's more detailed information here.

https://github.com/motioneye-project/motioneye/wiki/Install-In-Docker

JamBalaya56562, 2026-11-12:
shaynethiessen's answer is right: in a container, motionEye only sees the paths mounted into it, and anything else is written inside the container. Mount the disk with `-v /media/usbdisk:/media/usbdisk` (or under `volumes:` in docker-compose) and keep `/media/usbdisk` as the File Storage path. Closing as answered.

closed 2026-11-12 by JamBalaya56562