motioneye-project/motioneye #3029: won't start

max-nemirov, 2024-07-20:
I can’t figure it out, some kind of startup error

maxim@SVAROG ~ $ python3 -V
Python 3.10.12
maxim@SVAROG ~ $ sudo systemctl status motioneye
[sudo] пароль для maxim:
× motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2024-07-20 20:02:50 MSK; 28min ago
   Main PID: 18799 (code=exited, status=1/FAILURE)
        CPU: 118ms

SVAROG meyectl[18799]:     from motioneye import meyectl, motionctl, settings, tasks, uploadservices, utils
SVAROG meyectl[18799]:   File "/usr/local/lib/python3.10/dist-packages/motioneye/motionctl.py", line 26, in <module>
 SVAROG meyectl[18799]:     from tornado.httpclient import AsyncHTTPClient, HTTPRequest
 SVAROG meyectl[18799]:   File "/usr/local/lib/python3.10/dist-packages/tornado/httpclient.py", line 50, in <module>
 SVAROG meyectl[18799]:     from tornado import gen, httputil, stack_context
 SVAROG meyectl[18799]:   File "/usr/local/lib/python3.10/dist-packages/tornado/httputil.py", line 107, in <module>
 SVAROG meyectl[18799]:     class HTTPHeaders(collections.MutableMapping):
 SVAROG meyectl[18799]: AttributeError: module 'collections' has no attribute 'MutableMapping'
 SVAROG systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
 SVAROG systemd[1]: motioneye.service: Failed with result 'exit-code'.

max-nemirov, 2024-07-21:
Solved this problem

sudo apt install motion

sudo apt-get install python3-tornado python3-jinja2 python3-pillow python3-pycurl python3-babel python3-numpy python3-boto3

sudo pip install motioneye.eo

sudo motioneye_init

MichaIng, 2024-07-21:
Which version of motionEye did you use? `motioneye.eo` is a different project, deprecated to my knowledge, as it has been merged here.

max-nemirov, 2024-07-21:
> Which version of motionEye did you use? `motioneye.eo` is a different project, deprecated to my knowledge, as it has been merged here.

0.43.1b1
0.43.1b2
These versions didn't work for me.
Only works motioneve.eo

MichaIng, 2024-07-21:
It is a mismatch between some Python module versions, the `tornado` package in particular. `class HTTPHeaders(collections.MutableMapping)` is wrong in Python 3.10, should be `class HTTPHeaders(collections.abc.MutableMapping)` instead. I guess it was a bug in the package, and `apt install python3-tornado` fixed it.

Generally, you can use `pip` to install latest compatible version of all Python modules:
```sh
pip3 install -U tornado
```

Aviatorpaal, 2025-01-16:
> It is a mismatch between some Python module versions, the `tornado` package in particular. `class HTTPHeaders(collections.MutableMapping)` is wrong in Python 3.10, should be `class HTTPHeaders(collections.abc.MutableMapping)` instead. I guess it was a bug in the package, and `apt install python3-tornado` fixed it.
> 
> Generally, you can use `pip` to install latest compatible version of all Python modules:
> 
> pip3 install -U tornado

This!

JamBalaya56562, 2026-11-04:
As MichaIng found, this came from an old tornado that still uses `collections.MutableMapping`, which Python 3.10 removed, and updating tornado fixed it for @Aviatorpaal too. A fresh install no longer has this problem: on Python 3.10, `pip install motioneye` gets 0.44.0 with tornado 6.5.10, which does not use it, and the server starts. Closing as completed.

closed 2026-11-04 by JamBalaya56562

MichaIng, 2026-11-04:
Yeah, due to regularly published vulnerabilities, we keep the minimum Tornado version requirement quite up-to-date anyway. So a motionEye update will pull a recent Tornado version almost certainly.