motioneye-project/motioneye #3031: Problems connecting Dahua IP cam to MotionEye

Hardfield-D, 2024-07-22:
I am trying to connect to a Dahua IP cam in MotionEye by using the rtsp://[Camera-IP]:554/cam/realmonitor?channel=1&subtype=0
sequence.
This is working fine in VLC (where I am asked for username and password) but it is not working at all in MotionEye (where I specify username and password in the camera config)

Additionally the cam is working fine from my QNAP surveilance station on same port 554.

Does anyone have experience in connecting Dahua cameras to MotionEye ?

Dahua IP camera type is DH-SD22404T-GN

zagrim, 2024-07-23:
Well, I have two Dahua IP cams, and they are working just fine using URL of that same form (`rtsp://x.x.x.x:554/cam/realmonitor?channel=1&subtype=1`), and I have not disabled authentication in cameras.

Please check first that the camera credentials are saved ok in the camera config file (`/etc/motioneye/camera-1.conf`, replace 1 with actual camera ID number in ME) (look for option named `netcam_userpass`).

Are you BTW using the pre-release of 0.43 or the older ones? I'm not sure but at least some "special character" encoding issues in some passwords have been fixed in the Python 3 based 0.43 version - might be relevant if you have not just letters A to Z and numbers in the camera password. And there's plenty of other reasons, too, to upgrade from the older Python 2 based versions :smile:

Hardfield-D, 2024-07-23:
Thanks for the quick response.
After uninstalling and newly installing the very latest version that you recommended, it's working like a charm.

So next step for me will be the configuration to automatically getting some video sequences when motion is detected.
So far I am only receiving periodically image files (every 5 minutes as I have specified that) but no video sequences.

Do you know some properly working settings for MotionEye for that ? I have set my camera resolution to 1280x1024 - suppose that is good enough for the beginning.

zagrim, 2024-07-24:
Great that the upgrade resolved that issue!

Good motion detection settings depend on the scene and the conditions (outdoor/indoor, how much non-interesting changes can occur like lighting changes, wind etc). What follows is some instructions written based on my experiences trying to tune a setup with rather difficult environmental conditions.

Start by making sure you have "Recording Mode" set to "Motion triggered" under "Movies". 

Then, 

1. make sure "Motion detection" section is enabled, 
2. under "Motion detection" set "Frame Change Threshold" to as close to zero as possible (but not zero, as I seem to recall there's an unresolved UI bug that effectively disables motion detection when the slider is at zero), 
3. set "Maximum Change Threshold" to zero, 
4. disable "Automatic Threshold Tuning", 
5. enable "Automatic Noise Detection" (generally I've not found to usable trying to manually adjust that), 
6. set "Light Switch Detection" quite high or even to 100% if your scene is not expected to have rapid lighting conditions changes, and
7. enable "Show Frame Changes" to see the number of changed pixels for each frame in the UI to get better idea on how things work. 
 
The frame change threshold and some other motion detection settings having a percentage value mean the percentage value of changed pixels frame-to-frame, so with 1280x1024 frames, 1000 (changed) pixels results roughly 0.08% of the frame, and 50000 pixels results nearly 4% (to have some examples). If you want to understand better how Motion uses the frame change threshold, read https://motion-project.github.io/motion_config.html#threshold

After that, you should start getting motion detected and can start tweaking the settings to only capture "the good stuff". For that, it might be good to start with 

1. enabling "Despeckle Filter",
2. set "Minimum Motion Frames" to a value that means something like a second or two with your frame rate (like, if your frame rate was 20, set this to something between 20 to 40) to filter out uninteresting short events, **if needed**,
4. if the recording seems to miss some frames where motion already existed but was not yet/anymore detected, set "Captured Before" and "Captured After" numbers to, say, to the value of your frame rate to get an extra second the in start and end of the recording,
5. set "Motion Gap" to zero unless you need to have a "cool off" period after a motion event, and maybe most importantly :smile: ,
6. adjust "Frame Change Threshold" in small steps to filter out insignificant small changes so that they do not trigger a motion event.

If you need to filter out some areas of the frame from motion detection, set up an "editable" mask if the areas are static, or try out "smart" mask if those areas change over time.

I think I've also earlier given someone similar instructions but can't recall anything that would help finding them again, so I hope that had all the important things covered :sweat_smile:

zagrim, 2024-07-24:
If you need more assistance on getting motion detection working ok, please open a new issue (please check first if any previous issues around the topic can help you).

closed 2024-07-24 by zagrim

Hardfield-D, 2024-07-24:
Thanks for Sharing that good Information . I suppose that I can get it managed with all those items you mentioned in here.Von meinem iPhone gesendet