motioneye-project/motioneye #3033: [Errno 13] Permission denied .keep

ops244, 2024-07-31:
hi, again about the issue from this issue https://github.com/motioneye-project/motioneye/issues/1810
you could have motion eye remove this one.keep?
the thing is that linux makes it system-wide and hides it, and because of this, motion eye cannot remove it

I talked to the creator of samba and he said that you can solve the problem and offered several options there
https://github.com/home-assistant/addons/issues/3707#issuecomment-2259508392

as-kholin, 2024-07-31:
Being fair - I'm not the creator of Samba (https://www.samba.org/) - I'm one individual who happened to be looking at the HomeAssistant plug-in that pulls in a minimal Samba install (and I'm not even a maintainer of that project).  I'm just an interested individual - I have no official role with Samba or HomeAssistant

I also did not say Samba could not be made to change that setting.  The Samba software itself CAN be told to expose dot files ([https://www.samba.org/samba/docs/current/man-html/smb.conf.5.html](Man smb.conf.5), setting is 'Hide dot files').  

What I said was that the Samba HomeAssistant plug-in has no option today to expose that option - and that I was not sure exposing the setting from HomeAssistants plug-in was the right answer.  However, I have no authority to say it will not happen, only to provide an opinion (which is why I deferred to the maintainers)

However, I do not believe my answer was incorrect. Few individuals interact with Samba configuration directly - they generally interact with another package that presents it - and few of those implementations then add the setting to change the default for how Sambahandles Linux hidden files, given how niche the use case is.

With that said, having motioneye have an option to not use a file that conflicts with Samba/Linux hidden files, an option to say that it is using Samba (to factor it in), or having it confirm the file does not exist by trying to delete it and catching the fail to address cases like this would not go amiss.  

But, as with HA, the decision goes to the developers and maintainers of MotionEye for if this is development they want to pursue, and if so, where it fits for prioritization.

MichaIng, 2024-07-31:
Why opening a new issue if there is one already?

closed 2024-07-31 by MichaIng