motioneye-project/motioneye #3035: Issue: How to split file when continuous recording is activated

rjcab, 2024-08-05:
Hello,

I am running motion on debian with continuous recording.
The files are really heavy, about 45Go for 7 days

Do you know how to reduce the size and for example have one file for one day.

Thanks

zagrim, 2024-08-06:
Does it help if you set "Maximum Movie Length" to, say, 86400 seconds (=24 hours) under "Movies"? IMO it should, but I'm not quite sure if continuous recording mode ignores some settings.

JamBalaya56562, 2026-11-12:
zagrim's answer is the way: Movies → Maximum Movie Length is written to Motion's `movie_max_time`, which starts a new movie file after that many seconds, also with Continuous Recording; 86400 gives one file a day. To make the files smaller, lower Movie Quality or the frame rate, or choose a more efficient movie format. Closing as answered.

closed 2026-11-12 by JamBalaya56562