motioneye-project/motioneye #3037: ssl peer certificate or ssh remote key was not okay on DietPi

Ryu945, 2024-08-06:
1) When I try to use the SFTP upload option on the MotionEye deployed on DietPi, it gives the error 

>ssl peer certificate or ssh remote key was not okay

I also noticed that Digest mode of camera feed causes the camera to stop working.  I am wondering if the two problems are related.

I know there is no problem with the server it is connecting to since it is a working server with other devices connected to it.

2)  I noticed that the MotionEye deployed on DietPi does not have the the same storage options as the unmaintained MotionEyeOS.  Is there any plans to fix this in the MotionEye code?

Oli1L, 2024-08-07:
Hi,
I have the same problem when using SFTP with MotionEye: 0.43.1b2, Motion: 0.43.1b2 on a Raspberry Pi 5 with Debian 12 64Bit (Bookworm).
The following error is displayed: 51, SSL peer certificate or SSH remote key was not OK.

All I can do is use the insecure FTP server. Pity.

Does anyone know a solution?

Thanks.

JamBalaya56562, 2026-11-12:
I tried the SFTP upload test against an OpenSSH server on Debian 12 and 13, with both the system's pycurl and the one from PyPI (libcurl 7.88.1 to 8.23.0 with libssh2), with and without the server in `known_hosts`: it passed every time. So the error seems to depend on the server; libcurl reports it when the server's host key is rejected. As there were no details since 2024, I'm closing this. If it still fails on the current version, please open a new issue with the SFTP server's software and `ssh -v` output from the motionEye host. Other storage options than motionEyeOS had would be a feature request of its own.

closed 2026-11-12 by JamBalaya56562