motioneye-project/motioneye #3039: scalability

omumarvaishya005, 2024-08-07:
guys i want to know max camera handling capacity .

Since im using it for my collage project.
will it be able to more than 500 cameras

zagrim, 2024-08-09:
Scalability of MotionEye is almost entirely dependent of the scalability of Motion which does all the heavy lifting under the covers (motion detection, video en/decoding). And how many cameras can one computer handle with Motion is again hugely dependent on the camera feeds (resolution + framerate) and the specs of that computer. More than 500 cameras sounds quite much to me, but surely you can experiment by adding just a handful of cameras and checking the CPU, GPU and memory utilisation during motion events, and then extrapolating those to get the specs needed for your target setup.

zagrim, 2024-08-09:
If you don't require motion detection (just recording and real-time viewing of the feeds), then it's more likely that you could get those 500+ camera feeds successfully managed on a beefy hardware. But in that case any NVR software would do, and I would assume software meant for just NVR would suit your needs better than MotionEye, which is designed as a light-weight motion detection capable service which is most often (based on the number of blogs and guides written about it) run on a Raspberry Pi with just one or two cameras.

omumarvaishya005, 2024-08-11:
i want to use motion eye for 2 purpose.
First one : Need to get live feed from each camera.
Second one : Need to store integrate my own AI Modules like motion detection, human intervention, face detection etc etc..
so can i use it like this ??

Architecture :
                                                              -------------------------Feed is coming------------------------------
                                                              |                                                                                         |
                                              If motion is detected                                                  if not detected (no further steps)
                                                              |
                                 Start recording until there is a motion 
                                                              |
                                 Try to store the recording in data bases for further use 
                                                              |
                                  further uses will be like people counting, face detection , car plate number detection blah blah blah 
                                  
My only question here is can we add our own AI modules here for motion detection and do you have a functionality to record videos where there is a motion??

I Appreciate the response.

zagrim, 2024-08-13:
If you wish to use something else for basic motion detection than Motion, I'd say you'd be using very little of what MotionEye has to offer, and on the other hand it is not very good fit for your needs. Of course you could just set ME to record continuously to some storage and then put your own modules somewhere where they could poll the storage for new video files and do whatever they do after that. But you cannot replace Motion with your own motion detection stuff directly in MotionEye.
MotionEye can only record videos based on motion detection by using Motion (hence the name... :smiley: ), it is not designed for general platform where any sort of motion detector could be attached.

So, I again suggest you to check out generic NVR software, since they can do the same you'd apparently be using from ME, and their UI probably scales more nicely to the large amount of cameras you wish to attach.

omumarvaishya005, 2024-08-25:
could you please drop the motion eye nvr github repo???

zagrim, 2024-08-25:
I spoke generally, about any NVR software that does just recording of video. First hit from duckduckgo search on "open source nvr software" gave plenty from which to pick one that likely fits your needs: https://medevel.com/os-nvr-solutions/ (some have some sort of motion/object detection but since you wanted to do that by yourself, pretty much any of the ones on that list should be suitable, I think.