motioneye-project/motioneye #3043: Why it reloads a cameras when we add a new camera??

omumarvaishya005, 2024-08-11:
When we add a new camera, instead of adding a new camera it just add new camera + reloads all others by doing this there are several issues .
1> May be we will loose streams and valuable inputs of that time.
2> Not so User friendly (ill try to fix these issues).

zagrim, 2024-08-13:
I already answered to this same thing here: #3041

Adding or reconfiguring cameras is not happening all the time, is it? Sure, with those 500+ cameras in your scenario that probably could happen more ofter than in the usual ME user scenario with just a handful of cameras, but there are reasons for how it works (which I explained in the earlier issue you opened) and there's not much that can be done about it.

Motion does these days (not sure when that feature was released) have a web control interface where it is possible to change config parameter values "live", but I'm not sure if those changes will really take effect without restarting it. But still we'd need to also persist the settings to file, so setting config parameters would be essentially doubled in code, which is not nice.

JamBalaya56562, 2026-11-13:
Closing as a duplicate of #3041, as @zagrim said: adding or changing a camera writes Motion's configuration, which Motion only reads when it starts, so motionEye restarts Motion and with it every camera. That is still how it works today.

closed 2026-11-13 by JamBalaya56562