motioneye-project/motioneye #3053: Why is x11 needed?

starbasessd, 2024-09-19:
Trying to install in Lite version to minimize system usage and X11 gets installed during "sudo motioneye_init". A lot of other un-needed software gets installed, too, just tryting to figure out how to lighten up the load.

zagrim, 2024-09-19:
What is needed is motion, v4l-utils, ffmpeg and curl, and those are installed by `motioneye_init` if they are not already present. If some of those depend on X11 on some specific Linux distro would be odd IMO (none of them have graphical user interface), but it all depends on the distribution. Care to share any details? By "Lite" do you mean Linux Lite, or Raspbian Lite or yet something else?

JamBalaya56562, 2026-11-04:
No X server or desktop gets installed, only client libraries. Simulating what `motioneye_init` installs (`apt-get install --no-install-recommends motion ffmpeg v4l-utils curl`) on Debian 12, Debian 13 and Ubuntu 24.04 lists packages like `libx11-6`, `libxcb1` and `x11-common`, but no `xserver` or `xorg` package. On Debian 12, they come from `libavdevice59`, a library of FFmpeg that motion and ffmpeg depend on, so motionEye cannot leave them out. Closing as answered; please reopen if your system does get a desktop from it.

closed 2026-11-04 by JamBalaya56562