motioneye-project/motioneye #3054: wget of still images not working the first time

mrneutron42, 2024-09-20:
For the last 3 years, I've been running MotioneyeOS (from June 6, 2020) in a Raspberry Pi Zero W with a RPi camera module attached via the CSI camera connector.
motionEye Version	0.42.1
Motion Version	4.2.2+gitUNKNOWN
OS Version	motionEyeOS 20200606
I've been using a simple bash script run by a CRON job entry to capture images, for my weewx weather station page.  
This is the line in the script that does the capture:
`wget http://192.168.7.90/picture/1/current/ -O /home/weewx/public_html/images/webcam.jpg`
It only calls for an image 1 time.
Doing this has worked great, once per minute, for 3 years.
(Note, that with MotioneyeOS you don't need the port :8765 notation in your wget call.)
This week, I loaded some other Raspberry Pi Zero boards (with attached RPi camera modules) loaded with 32-bit Bullseye RaspianOS from July 2024.  
motionEye Version	0.43.1b2
Motion Version	4.7.0
OS Version	Raspbian 11
I got Motioneye working in Bullseye and thought I had working Raspberry Pi cameras that were working as well as my 2020 vintage MotioneyeOS RaspberryPi cameras.  Nope!
I found that the same wget command line I was using with the MotioneyeOS RPi cameras, does NOT work the same with the current Motioneye release running inside Bullseye RaspbianOS.  
I found that the wget command doesn't grab a valid image from Motioneye, UNLESS you call wget, wait 2 seconds, and then call wget AGAIN!
It's like Motioneye puts the camera to sleep and it takes 2 wget calls to wake it up and get a valid image??
By experimenting, I found that calling wget 2 times with a 2 second delay between calls, wakes up the camera and obtains a valid image.
If I don't do this, I get a 0 byte .jpg file.
```
wget http://192.168.7.91:8765/picture/1/current/ -O /dev/null
sleep 2
wget http://192.168.7.91:8765/picture/1/current/ -O /home/weewx/public_html/images/webcam.jpg
```
Why is it now necessary to call wget 2 times to get a valid image from the camera with the latest release of Motioneye?
Is the current Motioneye putting the camera module to sleep and thus it is necessary to issue 2 wget calls to wake it up?
If so, is there a setting that will keep Motioneye from putting the camera to sleep?

Help appreciated,

mrneutron42, 2024-09-21:
The thing I'm seeing from watching the Motioneye GUI in a web browser, is that the camera image appears to time out and go off after about 1 minute, if you don't click on it with your mouse pointer, to "keep it awake".
I think that is the issue that is causing the wget command to be issued 2 times, to get an image.
The first wget wakes up the camera.  The second wget acquires a valid .jpg.
[image]

MichaIng, 2026-02-12:
No not yet. You can apply the `dev` branch like this:
```sh
sudo systemctl stop motioneye
sudo python3 -m pip install --force-reinstall --no-deps 'https://github.com/motioneye-project/motioneye/archive/dev.tar.gz'
sudo systemctl start motioneye
```