motioneye-project/motioneye #3056: libcamera works, but no video from unicam

Ahndalu, 2024-09-27:
I have a Rasperry Pi 3 A + running a lite bookworm OS, debian release 12.  It has a ribbon cable type tnba2055 camera which is a Rapsberry Pi 'brand' camera.
The libcamera hello routine will take photos fine.  The MotionEye GUI through the web browser works ok.  When I add a camera, there's only one type it has for me available to add.  Then I add it and it gives a grey box with no video in it.  Troubleshooting has seemed to suggest libcamera stack doesn't work with MotionEye?   MotionEye version is 0.43.1b2  Linux kernel 6.6.47+rpt-rpi-v8 aarch64
[image]   I had great trouble with the MotionEyeOS and so went to just a lite RPiOS.  What should my troubleshooting steps be to get this camera to work with this Pi ?  thank you

simon-bonin, 2024-09-29:
I concur, cannot get libcamera to work with motioneye 43.1b2 on raspberry pi zero w bookworm. Have tried a lot of outdated how-tos and nothing worked. MotioneyeOS 2020 still works but would prefer to have access to full debian and motioneyeOS is deprecated now anyway... 

Will report if I find a way to make it work.

Ahndalu, 2024-09-29:
Do you know a path to a "working" MotionEye with my hardware?  (RaPi 3 A+, ribbon cable camera) Any "working" is fine - I don't care about full-featured or not.  I spent a lot of effort trying to make MOtionEyeOS work on the 3 A+, and that failed.  The best I've gotten is the present situation where 'lite' bookworm plus the app MotionEye will run - but no video in the added camera, and lib-camera-hello works.  Every other setup attempted has had a worse result.

Ahndalu, 2024-09-29:
It looks like the request for this fix has had an open issue.   https://github.com/motioneye-project/motioneye/pull/2765

simon-bonin, 2024-09-29:
If you just want a CCTV feed, you can use Motion only with bookworm on raspberry pi, then have a central motioneye server that captures the feed and does the motion detection and recording. Usefull if you have mutliple cameras.

I'm curious as of why MotioneyeOS doesn't work on your Rpi 3, which raspberry pi camera do you use? Mine (V2.1) works ok.

Matthew1471, 2024-12-04:
> > libcamera stack doesn't work with MotionEye?
> 
> Try the instructions in this comment: [#2812 (comment)](https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-1817391735)

They were incomplete but mine I've used repeatedly from a blank install
https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345

JamBalaya56562, 2026-11-01:
As Marijn0 pointed out, the camera works with current motionEye through `libcamerify`, following the steps in #2812; Matthew1471 posted a complete version there: https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345. Also checked on a Raspberry Pi 5 with a Camera Module 3 and motion 4.7.1. Native libcamera support is tracked in #2812, so closing this as a duplicate of it. If the camera still does not work with these steps, please reopen.

closed 2026-11-01 by JamBalaya56562