motioneye-project/motioneye #3057: Update  Wiki for raspbian

heuzef, 2024-10-01:
[This tutorial](https://raspberrytips.fr/installer-motioneye-sur-raspberry-pi/) is clearly most update than the [official wiki doc](https://github.com/motioneye-project/motioneye/wiki/Install-on-Raspbian) ?

MichaIng, 2024-10-03:
This one works for all systems: https://github.com/motioneye-project/motioneye?tab=readme-ov-file#installation
But you are right, someone needs to find time to cleanup the wiki. Actually I think most of those individual pages can be just removed.

MartinNK, 2024-11-01:
any special instruction for raspberry pi install on newest bookworm images? Any problems with new wayland insted of x11?

Matthew1471, 2024-12-05:
Please note it's now "Raspberry Pi OS".. if I recall correctly they renamed because "Raspian" sounded like a slur.

https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345

JamBalaya56562, 2026-10-28:
The per-distribution installation pages have been removed from the wiki (December 2025). There is now a single install method for all distributions in the README: https://github.com/motioneye-project/motioneye#installation. Closing, as the page this refers to no longer exists. If the README instructions do not work for you, please open a new issue.

closed 2026-10-28 by JamBalaya56562