motioneye-project/motioneye #3059: Direct installation on ubantu 22.04(jammy)

Omumarvaishya, 2024-10-07:
Hello guys i wanted some sort of customization on motioneye.
for that i needed to change some features and add new features 
how can i install it locally and run code to make changes according
to my need?? do we have and documentation for this??

robvdl, 2024-10-19:
If you clone it with git and change into the directory, then "pip install ." will do that.

robvdl, 2024-10-19:
Sorry I can't for the life of me recommends "sudo pip install" but the installation recommends that, I already raised that as #3066

MichaIng, 2024-10-19:
Since it is all Python scripts, you could also modify them after regular install. Of course an update would overwrite it, but that is the same with installing from a local clone, unless you raise the version string.

JamBalaya56562, 2026-11-05:
As answered above, clone the repository and run `pip install .` in it. To change the code, `pip install -e .` in a venv is more convenient, as the server then runs the files of your clone. Checked on `dev` with Python 3.12 in a container, not on Ubuntu 22.04 itself: after editing a file and starting `meyectl startserver`, the server ran the edited code. Restart it after each change. Closing as answered.

closed 2026-11-05 by JamBalaya56562