motioneye-project/motioneye #3066: Why do the installation instructions encourage "sudo pip install"

robvdl, 2024-10-19:
First of all, I personally don't run this project, but I am a Python developer and a friend came across it and installed it.

I was a bit shocked to see "sudo pip install" instructions, especially because that is going away in the future.

https://stackoverflow.com/questions/61452582/why-is-using-sudo-pip-a-bad-idea

Why are we encouraging this sort of installation? Why can't we recommend users to create a virtualenv instead and teach good practices?

I don't want to come across negative, but "sudo pip install" is always a bad idea.

MichaIng, 2024-10-19:
motionEye has a low number of dependencies without any specific version requirements. So IMO a venv/virtualenv does not add any benefit. If anyone is running into any conflicts with the system-wide installation, I would be interested to know, since it must be a very rare and special case.

However, you are right that Python generally moves towards venv only, recet Debian and Ubuntu install flag files with their distro packages which prevent `pip` from installing system-wide, without adding another flag or config option. So yes, we should switch to venv in our install instructions.

Please open a PR for the readme, if someone finds time. Also the install tests should be updated accordingly.

robvdl, 2024-10-19:
I dunno if I agree with the "does not add any benefits" because my friend did run into conflicts. I'll have a look at it when I can, but I'm going to try zoneminder with that setup instead.

There has to be a better way to install it. But I thought that we got over that "sudo pip install" thing years ago, almost nobody does that anymore.

MichaIng, 2024-10-19:
Did you even read the second half of my comment?

robvdl, 2024-10-19:
You're taking about the `--user` flag?

MichaIng, 2024-10-19:
No, that is still prevented on Debian and Ubuntu. This part of my comment is what I mean:

> So yes, we should switch to venv in our install instructions.
> 
> Please open a PR for the readme, if someone finds time. Also the install tests should be updated accordingly.

It is one more package `python3-venv` and two more commands to set it up and load it. The problem is that the systemd service needs to be changed as well. Not yet sure how to do this best. We would need to detect the venv in the `linux_init` script, if present, and in that case load it and get the `meyectl` path accordingly.

robvdl, 2024-10-19:
I'll have to have a good read of the code first to get an understanding of the installation, it looks like motioneye_init.py invokes a shell script which makes me wonder what the point of that is and why not just run the shell script directly. But there may have been a reason.

Makes me wonder "who not just `make install`" when there already is a Makefile there in the project root.

MichaIng, 2024-10-19:
Here is the shell script and systemd unit: https://github.com/motioneye-project/motioneye/tree/dev/motioneye/extra

I was wondering as well what the Python script purpose is. At least it is installed into PATH. However now it comes in handy, since it assures the script is executed within the venv, so that we can know the location of `meyectl`.

closed 2026-10-31 by Marijn0