motioneye-project/motioneye #3067: Wiki installation instructions are still for Python 2

robvdl, 2024-10-19:
The README.md is for Python 3, but Wiki is for Python 2.

Not sure why there are installation instructions in two places, but I imagine this project has moved past Python 2 if the README.md mentions Python 3.

MichaIng, 2024-10-19:
Yes, the wiki instructions IMO can be removed.

robvdl, 2026-02-22:
> Right, I just removed them 😅. If someone wants to add specific package instructions for RPM, APK, or other distro package managers, I'd prefer them right in `README.md`. This shouldn't be more than one line per type of distro. Not considering other implementations, like the AUR package. Out of scope to document them here, and IMO basically obsolete.

It's been a long time ago since I raised this ticket so I can't even remember where I used this project.

One thing I notcied though quickly scanning over README.md is something that is blocked in newer Ubuntu versions and other Linux distros will follow suit "sudo pip intsall".

I see "sudo python3 -m pip install" which is the same.

```
sudo systemctl stop motioneye
sudo python3 -m pip install --upgrade --pre motioneye
sudo systemctl start motioneye
```

Should really encourage creating a virtualenv in /opt and avoid sudo pip install because future Ubuntu blocks it.

I once brought this up at work "but what about the python docker image?" it turns out the Python in docker images is modified to allow root pip installs still.

MichaIng, 2026-02-22:
This is why step 2 of the install instructions contain a section to add the `break-system-packages` flag.

But it is still right that we need to switch to `venv`, as even with `break-system-packages` there are issues with recent `pip`: If a Python module is installed via distro package manager (e.g. `apt install python3-cryptography`), and the same package is installed via `pip install cryptography` on system level, `pip` now tries to remove the instance from the distro package manager, which of course fails. In the past, it just ignored these instances for good, since `/usr/local/lib` anyway has precedence over `/usr/lib`.

Here is an open issue about hat, PRs are welcome: https://github.com/motioneye-project/motioneye/issues/3066

In DietPi, we currently do this:
```sh
python3 -m venv --upgrade-deps /opt/motioneye
/opt/motioneye/bin/pip3 install -U motioneye
/opt/motioneye/bin/motioneye_init
sed -i 's|ExecStart=.*/meyectl|ExecStart=/opt/motioneye/bin/meyectl|' /etc/systemd/system/motioneye.service
```
I am wondering whether there is a sane way to autodetect the correct `meyectl` path in https://github.com/motioneye-project/motioneye/blob/dev/motioneye/extra/linux_init:
- If in `PATH`, that should be fine. `/opt/motioneye/bin/activate` should achieve this, and `motioneye_init` should preserve it when calling `linux_init`.
- If not in `PATH`, maybe `motioneye_init` executable location can be obtained, and `meyectl` should be in the same directory. If difficult to obtain within `linux_init`, `motioneye_init` can pass it as argument. A new option for `linux_init` to specify the `meyectl` path makes sense anyway. That would then take precedence over autodetection.
- It could be derived from the `linux_init` path as well, if the environment follows standard structure:
  ```sh
  echo "$(realpath "$(dirpath "$BASH_SOURCE")/../../../bin/meyectl")"
  ```