motioneye-project/motioneye #3072: Reolink camera works with 0.42.1 but not with 0.43.1b2

rolfhub, 2024-11-01:
Hi,
thanks for keeping this project alive and for porting it to Python3 👏!

I recognized that my Reolink camera (Reolink Duo Floodlight WiFi) works with 0.42.1 (Debian10, motion 4.1.1) but not with 0.43.1b2 (Debian12, motion 4.7.0) both running in Docker on x86_64. The camera is configured as "Network Camera" with RTSP URL: rtsp://192.168.178.60:554/h264Preview_01_main

The error message is: "ERROR: could not start mjpg client for camera id 1: not enabled or not local"

Is there anything that could be configured to enable this camera?

My two other cameras (Hikvision) are working correctly with both versions of motioneye!

rolfhub, 2024-11-02:
I could fix my problem by changing the baseimage in the [Dockerfile](https://github.com/motioneye-project/motioneye/blob/main/docker/Dockerfile) from debian:bookworm-slim to **debian:bookworm**.

JamBalaya56562, 2026-11-12:
Thanks for reporting back that it works with the full `debian:bookworm` base. The image is built on `debian:trixie-slim` today. `mp4:h264_omx` asks for an OMX hardware encoder, as on a Raspberry Pi, while `mp4`, as on your Hikvision cameras, encodes in software. Closing as solved; please open a new issue if the camera still fails with the current image.

closed 2026-11-12 by JamBalaya56562