motioneye-project/motioneye #3073: MotionEye Hub - An error occurred, refreshing is recommended

Exar50, 2024-11-06:
I set up a hub running bookworm on a Pi4.  No camera is attached to the hub.  I added remote MotionEye cameras and everything looked good.  Then a couple days later I get "An error occurred.  Refreshing is recommended." and the previously set up cameras are gone.  Couldn't figure it out, so I started from scratch, reloading the OS in the hub and setting everything back up.  It all looked good.  Then a day later I get the same error.

I don't think it's a memory error, I don't have any file saving turned on.  I tried looking at logs but I cannot find any, don't know what's up with that.

Suggestions?

Exar50, 2024-11-15:
I may have found the problem.  I just tried accessing the hub and it said "An error occurred. Refreshing is recommended."  I then checked the individual cameras and 2 of them were not working.  I rebooted them and they came back up.  I then went back to the hub and it worked fine.  So the issue may have been caused by the hub not able to access one of the cameras.

hakova, 2024-11-27:
> I may have found the problem. I just tried accessing the hub and it said "An error occurred. Refreshing is recommended." I then checked the individual cameras and 2 of them were not working. I [...]

Thank you for this posting. This should be the first item on the list for troubleshooting, rebooting the cameras, especially the Raspberry Pi ones. Your message saved me from hours of frustrating work, thank you!

Matthew1471, 2024-11-27:
To add a bit of technical detail to this issue (as I too experienced it) here is the technical output when a camera error occurs (`journalctl -u motioneye.service`):

```

Nov 27 22:52:37 MotionEye meyectl[497]:    ERROR: HTTP 599: Connection timed out after 10001 milliseconds
Nov 27 22:52:37 MotionEye meyectl[497]: Traceback (most recent call last):
Nov 27 22:52:37 MotionEye meyectl[497]:   File "/usr/lib/python3/dist-packages/tornado/web.py", line 1713, in _execute
Nov 27 22:52:37 MotionEye meyectl[497]:     result = await result
Nov 27 22:52:37 MotionEye meyectl[497]:              ^^^^^^^^^^^^
Nov 27 22:52:37 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/handlers/config.py", line 60, in get
Nov 27 22:52:37 MotionEye meyectl[497]:     await self.list()
Nov 27 22:52:37 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/handlers/config.py", line 505, in list
Nov 27 22:52:37 MotionEye meyectl[497]:     resp = await remote.get_config(local_config)
Nov 27 22:52:37 MotionEye meyectl[497]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 27 22:52:37 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/remote.py", line 214, in get_config
Nov 27 22:52:37 MotionEye meyectl[497]:     response = await _send_request(request)
Nov 27 22:52:37 MotionEye meyectl[497]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 27 22:52:37 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/remote.py", line 79, in _send_request
Nov 27 22:52:37 MotionEye meyectl[497]:     response = await AsyncHTTPClient().fetch(request, raise_error=False)
Nov 27 22:52:37 MotionEye meyectl[497]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 27 22:52:37 MotionEye meyectl[497]: tornado.curl_httpclient.CurlError: HTTP 599: Connection timed out after 10001 milliseconds
Nov 27 22:52:37 MotionEye meyectl[497]:    ERROR: 500 GET /config/list/?_=1732747947181&_username=admin&_signature=XXX (192.168.X.X) 10579.86ms
Nov 27 22:52:50 MotionEye meyectl[497]:    ERROR: HTTP 599: Failed to connect to Camera6.XXX port 80 after 2066 ms: Couldn't connect to server
Nov 27 22:52:50 MotionEye meyectl[497]: Traceback (most recent call last):
Nov 27 22:52:50 MotionEye meyectl[497]:   File "/usr/lib/python3/dist-packages/tornado/web.py", line 1713, in _execute
Nov 27 22:52:50 MotionEye meyectl[497]:     result = await result
Nov 27 22:52:50 MotionEye meyectl[497]:              ^^^^^^^^^^^^
Nov 27 22:52:50 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/handlers/config.py", line 60, in get
Nov 27 22:52:50 MotionEye meyectl[497]:     await self.list()
Nov 27 22:52:50 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/handlers/config.py", line 505, in list
Nov 27 22:52:50 MotionEye meyectl[497]:     resp = await remote.get_config(local_config)
Nov 27 22:52:50 MotionEye meyectl[497]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 27 22:52:50 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/remote.py", line 214, in get_config
Nov 27 22:52:50 MotionEye meyectl[497]:     response = await _send_request(request)
Nov 27 22:52:50 MotionEye meyectl[497]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 27 22:52:50 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/remote.py", line 79, in _send_request
Nov 27 22:52:50 MotionEye meyectl[497]:     response = await AsyncHTTPClient().fetch(request, raise_error=False)
Nov 27 22:52:50 MotionEye meyectl[497]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 27 22:52:50 MotionEye meyectl[497]: tornado.curl_httpclient.CurlError: HTTP 599: Failed to connect to Camera6.XXX port 80 after 2066 ms: Couldn't connect to server
Nov 27 22:52:50 MotionEye meyectl[497]:    ERROR: 500 GET /config/list/?_=1732747967978&_username=admin&_signature=ea128d9b987f3b1876f3b8992a12a94d8fcaf396 (192.168.X.X) 2341.16ms
Nov 27 22:55:03 MotionEye meyectl[497]:    ERROR: HTTP 599: Failed to connect to Camera6.XXX port 80 after 3051 ms: Couldn't connect to server
Nov 27 22:55:03 MotionEye meyectl[497]: Traceback (most recent call last):
Nov 27 22:55:03 MotionEye meyectl[497]:   File "/usr/lib/python3/dist-packages/tornado/web.py", line 1713, in _execute
Nov 27 22:55:03 MotionEye meyectl[497]:     result = await result
Nov 27 22:55:03 MotionEye meyectl[497]:              ^^^^^^^^^^^^
Nov 27 22:55:03 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/handlers/config.py", line 60, in get
Nov 27 22:55:03 MotionEye meyectl[497]:     await self.list()
Nov 27 22:55:03 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/handlers/config.py", line 505, in list
Nov 27 22:55:03 MotionEye meyectl[497]:     resp = await remote.get_config(local_config)
Nov 27 22:55:03 MotionEye meyectl[497]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 27 22:55:03 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/remote.py", line 214, in get_config
Nov 27 22:55:03 MotionEye meyectl[497]:     response = await _send_request(request)
Nov 27 22:55:03 MotionEye meyectl[497]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Nov 27 22:55:03 MotionEye meyectl[497]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/remote.py", line 79, in _send_request
[... cut ...]
            if decoded['error'] == 'unauthorized':
                response.error = 'Authentication Error'
            elif decoded['error']:
                response.error = decoded['error']
    except Exception as e:
        logging.error(f"_send_request: {e}")
        response = HTTPResponse(request, 500)
        response.error = e

    return response
```

Then `service motioneye restart`  gets you working again.

Not suggesting this is a rugged amazing patch.. but just a workaround bodge till someone properly looks at this :)

Exar50, 2024-11-28:
Thanks everyone for their feedback, and thank you Matthew1471 for your extra work on this.  I'm closing this issue.

closed 2024-11-28 by Exar50