motioneye-project/motioneye #3074: Error in Motioneye Daemon

AnthonyBSong, 2024-11-16:
Currently getting the following error when running systemctl status motioneye after installing it using the instructions for Raspberry Pi OS (buster): [https://github.com/motioneye-project/motioneye/wiki/Install-On-Raspbian].

Does anyone know what I am forgetting to install or which step I may have skipped which is causing this issue?

Error/fail below:
`× motioneye.service - motionEye Server
     Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sat 2024-11-16 17:57:48 EST; 2s ago
   Duration: 132ms
    Process: 3339 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
   Main PID: 3339 (code=exited, status=1/FAILURE)
        CPU: 131ms

Nov 16 17:57:48 raspberrypi systemd[1]: Started motioneye.service - motionEye Server.
Nov 16 17:57:48 raspberrypi meyectl[3339]: Traceback (most recent call last):
Nov 16 17:57:48 raspberrypi meyectl[3339]:   File "/usr/local/bin/meyectl", line 5, in <module>
Nov 16 17:57:48 raspberrypi meyectl[3339]:     from motioneye.meyectl import main
Nov 16 17:57:48 raspberrypi meyectl[3339]:   File "/usr/local/lib/python3.11/dist-packages/motioneye/meyectl.py", line 28, in <module>
Nov 16 17:57:48 raspberrypi meyectl[3339]:     import settings
Nov 16 17:57:48 raspberrypi meyectl[3339]: ModuleNotFoundError: No module named 'settings'
Nov 16 17:57:48 raspberrypi systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
Nov 16 17:57:48 raspberrypi systemd[1]: motioneye.service: Failed with result 'exit-code'.
`

Matthew1471, 2024-12-04:
Try these steps : https://github.com/motioneye-project/motioneye/issues/2812#issuecomment-2506840345

JamBalaya56562, 2026-11-12:
`import settings` in `meyectl.py` is the code of motionEye 0.42.1 or older, for Python 2, which the old wiki page installs; it does not run on Python 3.11. Please follow the [installation steps in the README](https://github.com/motioneye-project/motioneye#installation) instead, which install the current version into a venv. Closing this; please open a new issue if it fails that way.

closed 2026-11-12 by JamBalaya56562